keirf/greaseweazle #373: [BAD SECTOR] Thoughts

GeoKM, 2023-11-01:
Hi keirf, I can see the benefits of this in certain situations, but is there an option to switch this behavior off? 

ie. for 1541 often we want to keep the bad sectors etc in their true representation within the flux (scp) image.

In the future you may want to know that a file has been corrupted, and if the blocks now have good CRC's etc they might be slipped into a file as good data. Or obviously, as you would know, many C64 protections (especially the simple ones) rely on finding a specific block error in a certain location.

I've just started imaging all of my C128 & C64 disks as flux SCP's, granted 99% of my disks are not protected, but I do want to know that bad blocks are bad blocks.

Prehaps, keep the bad sector flux data in the SCP by default, and provide an option to replace it with [BAD SECTOR] on convert or write?

8-)

Thanks for a great device BTW 8-)

_Originally posted by @GeoKM in https://github.com/keirf/greaseweazle/discussions/264#discussioncomment-7442158_

keirf, 2023-11-01:
If you dump to raw flux then no conversion happens anyway. So for example you can validate against a --format during gw read, but if you also specify --raw then the exact raw flux coming off disk is what will be saved into your new image file.

GeoKM, 2023-11-06:
Yep Thanks Keir.Almost finished my 5.25’s 😁Sent from my iPhone