keirf/greaseweazle #374: 0 flux reads

josegoicoechea, 2023-11-10:
Hi
I'm hoping someone can shed some light on a an issue I'm having.
I can seem to get any flux reads happening. 
Here are some examples 
```
gw read blah.adf
Reading c=0-79:h=0-1 revs=1.1
Format amiga.amigados
T0.0: AmigaDOS (0/11 sectors) from Raw Flux (0 flux in 0.00ms)
T0.0: AmigaDOS (0/11 sectors) from Raw Flux (0 flux in 0.00ms) (Retry #1.1)
T0.0: AmigaDOS (0/11 sectors) from Raw Flux (0 flux in 0.00ms) (Retry #1.2)
T0.0: AmigaDOS (0/11 sectors) from Raw Flux (0 flux in 0.00ms) (Retry #1.3)
T0.0: Giving up: 11 sectors missing
T0.1: AmigaDOS (0/11 sectors) from Raw Flux (0 flux in 0.00ms)
T0.1: AmigaDOS (0/11 sectors) from Raw Flux (0 flux in 0.00ms) (Retry #1.1)
```
gw read blah.hfe::bitrate=500
Reading c=0-81:h=0-1 revs=3
T0.0: Raw Flux (0 flux in 0.00ms)
T0.1: Raw Flux (0 flux in 0.00ms)
T1.0: Raw Flux (0 flux in 0.00ms)
T1.1: Raw Flux (0 flux in 0.00ms)

I've tried the 6 different floppy drives and although the drive goes through each sector it just does not seem to read.
I have also tried different computers and various floppy disks.

Any ideas would be great?

keirf, 2023-11-10:
What type of Greaseweazle? Have you tried a different cable?

j-goi, 2023-11-10:
It's a v4 board.
The drive cable is the only thing I haven't swapped.
Just need to get a hold of another one.

pauldsmyth, 2023-11-10:
I also have this issue. Greaseweazel V4. Floppy cable purchased with the unit

keirf, 2023-11-10:
It's an unusual error because more typically the drive doesn't get selected (and you get a "no track 0" error) or it doesn't spin (and you get a "no index" error). Reading no flux whatsoever is more commonly a hardware issue. It could be cable, drive, or possibly the Greaseweazle. Usually the cable.

keirf, 2023-11-10:
@pauldsmyth since you are in UK, if you continue to have problems I can easily send you another V4 unit to try. And/or you can send me yours to test.

pauldsmyth, 2023-11-10:
Sure, sounds good. I've ordered a new floppy cable to test. This floppy drive hasn't been used for years so I wouldn't have been surprised if it failed. It did successfully read a disk and erase it

Chrysopidae, 2024-03-14:
Has there been a verdict on this? I have the same problem with a 5.25" full height Tandon TM100-2A drive & Greaseweazle F7 Plus. The GW is able to read and write (using the same cable) to a 3.5" drive and another TM100-2A. No problems, works well.

However, its TM100-2A "brother" from the same IBM computer shows only "0 flux". Multiple discs and terminator scenarios make no difference. Erase, clean and seek work, RPM shows good speed. The drives have different board REV numbers though.

Unfortunately I have no way of running the drive regularly. They were sold as working, but I only have the GW to check.