keirf/greaseweazle #375: DEC RX01 RT-11 images written with GW4 boot, but writing to them from RT-11 corrupts the directory

bc17101, 2023-11-14:
I am able to successfully take images from various sites and create bootable 8" RX01 RT-11 floppy disks with the Greaseweazle. When I try to update a file on the GW written disk after booting it, the directory is corrupted. if I take the GW written disk, boot from it and image copy it to another disk using RT-11, the final disk works fine and I can successfully write/update files on it. So it seems that the image data is correct since the final RT-11 copied disk works great, but perhaps something at the  lower flux level isn't quite right.

Does anyone have any idea what might be causing this problem?

keirf, 2023-11-22:
Take raw dumps (eg SCP) of the bad original disk and fixed copied disk, and share them.

keirf, 2023-11-22:
RX01 should be simple as it is just FM. But maybe there's a twist somewhere.

bc17101, 2023-11-27:
Hi Keirf,

I apologize for the delay, but I cannot recreate the issue I reported. It turns out my original RT-11 disk image was bad. I since have a new master image and when I create a disk with the GW I can now delete and add files to the GW generated disk at will with no issues. Please close this issue.

Thank you for trying to assist in a solution. I love your GW, it’s been a life saver!

Best regards,

Bill Caldwell

closed 2023-11-28 by keirf