keirf/greaseweazle #377: Writing .dim files in v1.16.1 is not possible

sdsnatcher73, 2023-11-23:
in v1.16.1 when I try to write a .dim image I get:
gw write A-Jax\ \(1989\)\(Konami\)\(Disk\ 1\ of\ 3\)\(Program\).dim
** FATAL ERROR:
Sector image requires a disk format to be specified

So I specify the format:
gw write --format pc98.2hd A-Jax\ \(1989\)\(Konami\)\(Disk\ 1\ of\ 3\)\(Program\).dim
Writing c=0-76:h=0-1
Image format pc98.2hd
** FATAL ERROR:
Cannot override image format with --format

Tried with v1.15.1:
gw write A-Jax\ \(1989\)\(Konami\)\(Disk\ 1\ of\ 3\)\(Program\).dim
Writing c=0-81:h=0-1
Image format pc98.2hd
T0.0: Writing Track (Flux: 166.7ms period, 183.3 ms total, Terminate at index)
T0.1: Writing Track (Flux: 166.7ms period, 183.3 ms total, Terminate at index)
T1.0: Writing Track (Flux: 166.7ms period, 183.3 ms total, Terminate at index)
T1.1: Writing Track (Flux: 166.7ms period, 183.3 ms total, Terminate at index)
...

keirf, 2023-11-23:
Can you attach or link to the dim file please.

sdsnatcher73, 2023-11-23:
[A-Jax (1989)(Konami)(Disk 1 of 3)(Program).dim.zip](https://github.com/keirf/greaseweazle/files/13447620/A-Jax.1989.Konami.Disk.1.of.3.Program.dim.zip)

keirf, 2023-11-23:
Thanks. DCP is probably broken too. Do you happen to know where I could find one of those for testing?

sdsnatcher73, 2023-11-23:
[Rouge no Densetsu (1996)(D.O.)(Disk 01 of 12)(Disk A)[Req Install].dcp.zip](https://github.com/keirf/greaseweazle/files/13447879/Rouge.no.Densetsu.1996.D.O.Disk.01.of.12.Disk.A.Req.Install.dcp.zip)

keirf, 2023-11-23:
Please can you try the following test build:
https://github.com/keirf/greaseweazle/suites/18439459802/artifacts/1069057816

You need to be logged into GitHub for the download link to work.

This should fix both DIM and DCP. If so, I will roll out a 1.16.2 release.

sdsnatcher73, 2023-11-23:
Seems to write, don't have the machine yet to test if it writes correctly (but no reason to suspect it does not). It is strange that it no longer writes the Image format:

Dev release:
*** TEST/PRE-RELEASE: 1.16.2.dev6+g5ad2981
*** Use these tools ONLY for test and development!!
Writing c=0-81:h=0-1
T0.0: Writing Track (Flux: 166.7ms period, 183.3 ms total, Terminate at index)
T0.1: Writing Track (Flux: 166.7ms period, 183.3 ms total, Terminate at index)
T1.0: Writing Track (Flux: 166.7ms period, 183.3 ms total, Terminate at index)

v1.15.1:
Writing c=0-81:h=0-1
**Image format pc98.2hd**
T0.0: Writing Track (Flux: 166.7ms period, 183.3 ms total, Terminate at index)
T0.1: Writing Track (Flux: 166.7ms period, 183.3 ms total, Terminate at index)
T1.0: Writing Track (Flux: 166.7ms period, 183.3 ms total, Terminate at index)

keirf, 2023-11-23:
I got rid of the name print while cleaning up. Most self-describing formats in Greaseweazle (EDSK, D88, ...) don't print this type of information. Is it useful?

sdsnatcher73, 2023-11-23:
Can be helpful if selection goes wrong, you could put it behind a --verbose...

keirf, 2023-11-23:
Okay I added it back.

keirf, 2023-11-23:
Rolling out v1.16.2

closed 2023-11-23 by keirf