keirf/greaseweazle #383: Invalid data in Flux output

Digitoxin1, 2023-12-21:
gw.exe appears to be creating some flux tracks with invalid data in the raw output.  Here is an example:

The section highlighted in blue is an OOB index block.  Right after this block, highlighted in yellow is two bytes of &H00 which, from my understanding, should not be there and is invalid data.  This only occurs rarely in some tracks, but dumping the disk multiple times will reproduce this condition in the same track.  This issue does not occur when dumping the same disk using a KryoFlux (I do not personally own a KryoFlux, so I'm relying on a reputable third party here).   

[image]

Digitoxin1, 2023-12-21:
Update:  If I do a raw write back to a disk and then create a new flux set from that disk, this issue does not exist in the second generation flux image.

keirf, 2023-12-21:
That's a flux which occurs exactly on the index mark. The code `00 00` simply means 0 sample clocks to the next flux reversal. Since flux isn't perfectly synchronised to the index, and in many track formats the index is where writes start and stop, it is perfectly possible to find a flux reversal appearing to sit perfectly on top of the index. At Greaseweazle resolution you would expect to see that at least 1 in 500 tracks.

Digitoxin1, 2023-12-21:
Got it.  So this is by design.

closed 2023-12-21 by Digitoxin1