keirf/greaseweazle #384: Supported disk image formats could be updated in the documentation and in the --help option

Yserra, 2023-12-21:
Hi Keir, 

The gw read --help command lists some supported file suffixes that are actually not supported with the read command: 

.a2r, .d88, .dcp, .dim, .fdi, .ipf, .td0

And the documentation lists some "write only" disk image formats that the read command does actually support:

.hdm, .xdf, .dsk, .edsk

[image]

keirf, 2023-12-27:
Yes this has turned into a mess. I also need to update the wiki with all new supported image types.

peoyli, 2024-03-29:
While (when) at it, how about some documentation about disk formats per physical media type ? This could be useful when having to guess the what format the actual content is.
I'm still working on my "guess-format" shell script, untouched since 25 Oct (and really deserving this version tag):
# $VER: gw-guess-format 0.1 Final Edition (yeah, really) Update 4 (25.10.2023)

That script uses a file containing the formats to use for the guessing. I have a "formats-all" which probably is incorrect (because of documentation and --help being incomplete) and a "formats" which contains a limited set of formats I want to try when guessing, simple enough:
`amiga.amigados adf
amiga.amigados_hd adf
mac.400 ima
mac.800 ima
ibm.720 img
ibm.1440 img`

Where I have the first part being added to the output filename (to keep it unique) and the second part as file extension.

keirf, 2024-08-30:
Fixed the wiki. Not sure about fixing `gw read --help`, it's a bit of a pain and the command fails early with a clear error message for the few unsupported formats.

closed 2024-08-30 by keirf