keirf/greaseweazle #386: Greaseweazle v4 read problem (0 flux in 0.00ms)

boohyaka, 2023-12-24:
Hi, I see there's another thread with the same symptoms, still thought it was better to open my own, hope that's OK.

My v4 board (bought preassembled from a known online shop) used to work fine. Haven't touched it for some time. Tried to use it again, said a firmware update was available, so I did and it updated to v1.5. 

gw info shows the board correctly.

When trying to read, the drive is actually working and the disk accessed, the GW green led lights up, but I'm only getting "Raw Flux (0 flux in 0.00ms)".

I tested several known good drives, several ribbon cables, several USB cables, with host tools 1.10 and 1.16.2, same results in every case.
Tried to "gw update --force" too, it works fine, but still same results. 

Any other suggestion? Is my GW toast? :(

Thanks a lot for the help and happy holidays.

[image]

[image]

keirf, 2023-12-24:
Well it doesn't sound good. What sort of ribbon and what sort of drive(s)? 

I'm going to assume you are US based? I can provide easier assistance via mail in the UK.

keirf, 2023-12-24:
When you say "gw update --force" by the way, is that to try a downgrade of the firmware? Because you can do that, for example: `gw update --force --tag v1.4`

boohyaka, 2023-12-24:
Hi, I am based in Switzerland. Only a channel to cross, a bit closer than across the pond :)

I've tested with various drives that I know are good: TEAC FD-55GFR (5.25" 1.2MB), TEAC FD-55BR (5.25" 360KB), Mitsumi D359M3D (3.5" 1.44MB)... I tested both in Shugart mode (DS0/1/2 and straight ribbon cable that I made, I built it myself but tested all pins with a multimeter in continuity mode so I'm pretty confident it's good.) and in PC-AT mode (DS1, twisted ribbon). . My goal was actually to do a 3-drives Shugart setup initially.

Tried in multiple drives configuration, but also each drive individually to find if something actually work, but same result every time.  RIght now I'm in the "simplest" setup, a 3.5 drive connected after the twist of a regular PC ribbon cable (drive A).

Great tip about the downgrade, thank you, I didn't know. Downgraded to 1.4 but unfortunately same result. Anything else I could try? Really appreciate the quick support btw!

Cheers

keirf, 2023-12-24:
Can I ship to Switzerland without the ridiculous hoops required sending to EU these days (EORI, IOSS, non-existent import tax thresholds, etc)?

It sounds like you know what you're doing so it's probably busted. You could I suppose put a scope or LA on the rdata pin?

boohyaka, 2023-12-24:
I've been exchanging with UK recently with no issues so I guess it should be fine? Trick as always is to send as gift, with a declared value (item+shipping) lower than 100 to avoid import taxes and usually it's no trouble.

I know what I'm doing in terms of IT, but I'm very much a beginner in electronics. I did purchase a scope in the hope of learning more though! With a little hand-holding and clear instructions I should be able to do what you need, if we can at least get a proper diagnostic, before shipping anything one way or the other.

Waiting on your next answer I'll try to give a look to the technical documentation to find the rdata pin.

boohyaka, 2023-12-24:
OK floppy connector pin 30.

keirf, 2023-12-24:
Okay another thing you could try is a continuity test from rdata pin at the floppy header to the pin at the MCU. You would find the pin numbers on the schematic, which is included in the Design Files page of the wiki. Actually it would suffice to place one probe at floppy header pin 30 (rdata) and sweep the other probe across all MCU pins listening for the continuity beep.

keirf, 2023-12-24:
In any case you can find my email address in my GitHub profile and send me your postal address. I have a few spare V4 boards and can send you one to try.

boohyaka, 2023-12-24:
OK scope on pin 30 of the floppy connector gives me *something* but I have no idea what I should be expecting haha

Did test continuity from pin 30 floppy header to all ARM chip pins, and the only continuity is to pin 38, which seems correct?

anything else? Will send you an email in the meantime as proposed, thank you.

keirf, 2023-12-24:
All falling edges on rdata count as a flux. And if this signal has continuity to the MCU then there should be flux recorded during gw read. So it seems the board is indeed broken. Perhaps it is the pullup resistor or perhaps it is the MCU.

boohyaka, 2023-12-24:
Received your answer by email - many thanks for the quick help again!!

keirf, 2024-01-09:
Replacement unit shipped and confirmed working. Hence we conclude there is a hardware fault with the original board.

closed 2024-01-09 by keirf