keirf/greaseweazle #395: Proposal for slightly more elaborate documentation

Yserra, 2024-01-07:
Hi Keir, 
I wish you an happy new year. 

Since I find this page https://github.com/keirf/greaseweazle/wiki/Getting-Started a bit frustrating, I've tried to write a little more comprehensive documentation. See the attachment. 

Feel free to take from it as much as you like. 

Please note: English is not my mother tongue (I'm French) and this text may contain language errors.

Hope it could be useful.

Best.
[GreaseweazleDoc.zip](https://github.com/keirf/greaseweazle/files/13854625/GreaseweazleDoc.zip)

keirf, 2024-01-08:
Wow, that's a lot better!

Yserra, 2024-01-08:
I have no idea. Actually, I wrote this with Microsoft Word and saved it as .html. I tried to convert the .docx to .md with Pandoc, but it was a bit of a mess (the tables, the images went away, the font was non-sense). Would you have any recommandation to share that would help me produce the format you are expecting?

keirf, 2024-01-08:
Thanks. I can do a lot more with the docx original file. For example pandoc produces halfway sensible Markdown which I can tweak.

keirf, 2024-01-08:
See what you think of the new Getting Started wiki page!
https://github.com/keirf/greaseweazle/wiki/Getting-Started

keirf, 2024-01-08:
Images currently missing. To be added :)

Yserra, 2024-01-08:
Thank you Keir, It's fantastic! I'm honored to be part of this great adventure :)

closed 2024-03-19 by keirf