keirf/greaseweazle #398: Feature Request: add track min/max option for custom output pins

iz8dwf, 2024-01-11:
Hi, for the old 8" disk drives that have a Low current (aka TG43) input and/or Read Filter input in read mode, it would be great if one could use an uncommitted output and specify on what track interval the pin should go low/high. For the purpose of the Low Current/Read Filter input, check the manual for the YE-180 drive for example. Python is not really one of the languages I know, otherwise I would try to propose a patch too :)
Thanks

keirf, 2024-01-11:
You do know you can work around this by dumping raw to Kryoflux stream files in two passes? Then you can analyse these raw files separately via gw convert. Something like
```
gw pin set
gw read --tracks=c=0-N name_00.0.raw
gw pin set
gw read --tracks=c=N+1-76 name_00.0.raw
gw convert name_00.0.raw [...]
```

iz8dwf, 2024-01-11:
Yes, I failed to mention the fact that of course one can do "things" in two different files, but then it doesn't seem to me that's so straightforward to come up with a single disk image.

keirf, 2024-03-19:
`--gen-tg43` has been added to `gw write`.  Beyond that, splitting the reads is the way to go. Read to Kryoflux STREAM files which are per-track files anyway. Then merging isn't necessary. The raw files can always be processed and/or transcoded later using `gw convert`.

closed 2024-03-19 by keirf