keirf/greaseweazle #402: Unable to write ADF files with verify left on

modo74, 2024-01-26:
Hello all, Newbie to all this but I have an issue when writing ADF files to disk. 

I have a Geaseweazle v4.1 (updated to firmware 1.5) and a "new" old stock Sony MPF920 3.5 floppy drive (looks mint condition with no signs of use and quiet in operation). Using FluxMyFluffyFloppy (v5.0.3). 

Was able to format and read disks fine but had trouble getting anything ADF file to write until I selected the "No verify" option. Here is what I get with the "No verify" option off:

C:\FluxMyFluffyFloppy-master>"C:\greaseweazle-1.16.2\gw.exe" write --retries=3 "I:\AmigaTestKit-1.21\AmigaTestKit.adf"
Writing c=0-79:h=0-1
Format amiga.amigados
T0.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T0.0: Writing Track (Verify Failure: Retry #1)
T0.0: Writing Track (Verify Failure: Retry #2)
T0.0: Writing Track (Verify Failure: Retry #3)
** FATAL ERROR:
Failed to verify Track 0.0

Enabling "no verify" and the disk appear to write ok but concerned there may an issue with the drives write quality. Is this normal behaviour, have I missed an option for this drive or does it sound like a bad drive?

Any ideas?

keirf, 2024-01-26:
If the drive can write, and it can read, then verified write should not be a problem. Have you tried other disks? Are you *sure* the write did actually write?

modo74, 2024-01-27:
Tried a variety of disk old and new including a box of new sealed HD disks. Same result for all. 

Disks are writing although I'm getting the occasional read error when I try them on my Amiga A1200, although that could be the age of the internal drive. Without verifying the write I'm not sure if the read errors are coming from the write or the read phase.

keirf, 2024-01-27:
If you're using HD disk be sure to cover the HD hole. Otherwise you can expect read errors on a high density drive (double density Amiga drives don't look for the HD hole).

modo74, 2024-01-27:
Perfect, that's got it working correctly now.

Many thanks

closed 2024-03-19 by keirf