keirf/greaseweazle #410: TrDos for ZxSpectrum.

Motohobiman, 2024-02-10:
Hello!
Very thx for this product greasrweasle.
Plz add TrDos ZxSpectrum -  namefile.trd ,for working in this project ,many people ex ussr used this operation system.

Yserra, 2024-02-15:
I think TrDos is already implemented.

gw read --format=zx.trdos.640 MyZXSpectrumDisk.img

Yserra, 2024-02-15:
I think it's already implemented. Try:

(for a double-side 80-track disk):
gw read --format=zx.trdos.640 MyTRDosDisk.img
mv MyTRDosDisk.img MyTRDosDisk.trd

(for a single side, 40-track disk, read from a 48TPI drive) 
gw read --format=zx.trdos.640 MyTRDosDisk.img --tracks="c=0-39:h=0"
mv MyTRDosDisk.img MyTRDosDisk.trd

(for a single side, 40-track disk, read from a 48TPI drive) 
gw read --format=zx.trdos.640 MyTRDosDisk.img --tracks="c=0-39:h=0"
mv MyTRDosDisk.img MyTRDosDisk.trd

(for a single side, 40-track disk, read from a 96TPI drive) 
gw read --format=zx.trdos.640 MyTRDosDisk.img --tracks="c=0-39:h=0:step=2"
mv MyTRDosDisk.img MyTRDosDisk.trd

It it doesn't work, you can ever convert .img disk images to .trd disk images thanks to the tool [HxC Floppy Emulator](https://github.com/jfdelnero/HxCFloppyEmulator/releases)

Motohobiman, 2024-02-21:
Im use https://desertsagesolutions.com/greaseweazlegui/ 
and please addon .trd profile 
[image]
HxC Floppy Emulator not comfort working with trd file

Jujan, 2024-10-25:
For anyone wondering how to write TR-DOS files:

(FluxMyFluffyFloppy doesnt recognise .TRD file, you have to rename to .DSK, which is still regular to use for TR-DOS disks)

I assume you own High Density drive with Double Density disk inserted.

It is crucial to write the file depending on what geometry it is formatted.

(If not sure, then write the disk using default settings:
gw.exe write --format zx.trdos.640 example.trd
Then insert disk to BetaDisk interface and do LIST - the geometry is reported in the upper right corner, abort upon errors)

If the disk is 40 tracks 1 side then use:
gw.exe write --format zx.trdos.640 --tracks=c=0-39:h=0:step=2 example.trd

If the disk is 40 tracks 2 sides then use:
gw.exe write --format zx.trdos.640 --tracks=c=0-39:step=2 example.trd

If the disk is 80 tracks 1 side then use:
gw.exe write --format zx.trdos.640 --tracks=c=0-79:h=0 example.trd

If the disk is 80 tracks 2 sides then use:
gw.exe write --format zx.trdos.640 example.trd

These settings ensure reliable write of TR-DOS disks on standard BetaDisk interface TR-DOS ROM.
Some ROMs may accept other settings, but there are too many to list them all. This will cover 99% use cases.