keirf/greaseweazle #413: Format td0 : not able to create a disk

sappas2, 2024-03-02:
Hello, i'm trying to make a boot c/pm disk for my epson qx-10,
a teledisk format exist in greaseweale, but i'm not ablle to create it until i'm using hcx floppy emulator software to export .scp file.
By the way, why to create .scp ?

http://www.vintagecomputer.net/fjkraan/comp/qx10/disklibrary.html
And particulary this file http://www.vintagecomputer.net/fjkraan/comp/qx10/diskimages/QX10CPM_MF.zip

keirf, 2024-03-03:
What happens when you try to write the td0 file (I assume CPM22M1.TD0) directly using gw tools? The file appears to be imported okay, as a 16 sectors-per-track 40-track 2-sided image.

There should be no need to do prior conversion to scp.

My guess, verification fails when you try to write the td0. When you convert to scp, the write is not verified (gw cannot do a verified write of scp). Hence the write appears to work, but doesn't really.

sappas2, 2024-03-03:
You're right it's verify failure, i've changed 3 times from floppy drive et 2 times the cable, maybe the card and a cold junction ? 
Powersupply for floppy are external, an atx one

keirf, 2024-03-03:
What sort of drive and what sort of disk? The image is double density and so will need a double density disk.

sappas2, 2024-03-03:
Chinon FZ502 cleaned head and rail reoil and JU-455  5(*2) Mitsubishi DS DD soft sectored, 40 tracks, never used.
Can i abort verification as --retry=0 i cannot retrieve if  i've seen it 
In GW 1.16.3 after reading TD0, why "Writing c=0-81:h=0-1" for 40 cylinders ?

keirf, 2024-03-03:
Okay, so both drives are double-density and 40-track as required. Are the drives known to work? Can you read an existing known good disk, if you have one?

You can specify `--no-verify` to disable verification.

The `Writing c=0-81:h=0-1` is misleading actually. This is the default range of tracks considered, however nothing will be written for tracks not defined in the TD0 file, so in fact only c=0-39 will be written and your drive won't get its heads stepped out of range.

sappas2, 2024-03-07:
I can't read an existing disk, i close this issue, many thanks

closed 2024-03-07 by sappas2