keirf/greaseweazle #415: 8" Floppy: extra empty tracks in the image

gotcha-z80, 2024-03-07:
Hello,

When I dump 8" floppy that have 77 tracks as SCP images, I get extra empty tracks. It's not always the same number. Usually, I get between 126 and 130 tracks in total.

The command I use for a single sided floppy is the following:
```
gw read --drive=1 --revs=3 --tracks=c=0-76:h=0 test.scp
```

I attach an example displayed with HxCFloppyEmulator

[image]

keirf, 2024-03-07:
There is confusion around how to represent a single-sided SCP image. You could try latest beta/snapshot of HxC software, or you can produce a "legacy" single-sided SCP image: `gw convert test.scp test_legacy.scp::legacy_ss`

keirf, 2024-03-07:
Yes, I believe we came to agreement some time ago. I'm not sure how often the non-beta download gets updated -- the beta is often the best one to go for.

closed 2024-03-07 by keirf