keirf/greaseweazle #416: PC disk types

AngelDevIndie, 2024-03-09:
Hello Keirf,
Thank you for your work. I'm dumping all the PC games I have. 

I think I have seen an error in the parameter that is passed to the program. It's not important, but I think there was a confusion with the numbers.

In _SCP.py_ the _DiskType_ values for PC are:
[image]

In the specifications for **SCP** files on the cbmstuff.com website:
[image]

So the correct parameter should be:
**ibmpc-360k**

I don't know if this is the case, but maybe you got confused.

Thanks again, and I hope I helped you.

Best regards.

keirf, 2024-03-09:
Yes, that's a typo, thanks for the report!

closed 2024-03-09 by keirf

FrankieTheFluff, 2024-08-25:
Hi, could you please also check the added atarist-dshd and atarist-sshd?
https://www.cbmstuff.com/downloads/scp/scp_image_specs.txt

Greetings Frank

commit 49522e12 (2024-03-09): scp: Fix disktype typo: ibmpc-320k -> ibmpc-360k

Thanks to AngelDevIndie for the report.

Fixes #416