keirf/greaseweazle #418: Non-zero-indexed track numbers not handled correctly

Rua, 2024-03-18:
The Philips P2000 uses a special version of the Amstrad CPC format, in which the tracks are numbered from 1 rather than from 0. This means that when you seek physically to track 0, you see sectors whose metadata says they belong to track 1. In image files, these are still presented as *physical* track 0, but the sector data in that track is labelled as track 1. I've attached an example DSK image of the P2000 Disk Basic where this can be seen, and a screenshot of HxE Floppy Emulator showing the aberrant track numbering.

[image]
[P2000 Disk Basic.dsk.zip](https://github.com/keirf/greaseweazle/files/14640347/P2000.Disk.Basic.dsk.zip)

When writing such an image with the Greaseweazle, the GW software seems to ignore the physical numbering of each track. The sectors whose data says track 1 are written to physical track 1, thereby leaving track 0 empty (or with whatever data was there previously). This creates a floppy that is not readable by the P2000.

HxE Floppy Emulator handles this format correctly, and can convert it to a raw flux format. The Greaseweazle is then able to write that raw flux file with no issues. GW only goes wrong when it's given a DSK file. The issue may also exist with other formats, but I have only encountered it with this one and there may not actually be any other machines with this odd kind of track numbering anyway.

keirf, 2024-03-18:
So this is a straightforward `gw write` of the attached DSK file, and the tracks are incorrectly offset?

Rua, 2024-03-18:
Yes, pretty much. In the image, track 0 contains sectors whose track number is 1. The GW then writes those to physical track 1, and not to physical track 0 where they should be.

keirf, 2024-03-18:
You're not trying to specify a `--format` are you? Never do that with CPC DSK files, as they are self describing.

Rua, 2024-03-18:
Ah, I was using Greaseweazle tools 1.13 still. I'll see if I can still reproduce it with 1.16.3.

closed 2024-05-31 by keirf