keirf/greaseweazle #420: Is a Blue Pill definitely fake if it fails Blinky?

MeteorSlime, 2024-03-22:
I bought a Blue Pill: https://www.ebay.com/itm/176265114468
On the Chip it says STM32f103c8t6
And flashed Blinky onto it, the LED didn't Blink.
I made my own Blinky with the Arduino IDE:
`/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  https://www.arduino.cc/en/Tutorial/BuiltInExamples/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(PC13, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(PC13, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(PC13, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}
`
I uploaded it and it and the LED blinked.
Does that mean my Blue Pill is fake or not?

keirf, 2024-03-23:
It does mean it's fake. They will perform many operations correctly, but more complex stuff involving certain peripherals, and DMA, and interrupts may behave oddly. If all you want to do is blink an LED then they're certainly fine. ;)

closed 2024-03-23 by keirf

MeteorSlime, 2024-03-24:
> It does mean it's fake. They will perform many operations correctly, but more complex stuff involving certain peripherals, and DMA, and interrupts may behave oddly. If all you want to do is blink an LED then they're certainly fine. ;)

Thank you very much for the answer.
Do you know a genuine source of Blue Pills?

keirf, 2024-03-24:
I actually think they're all fake. Going down that avenue I would expect to replace the fake MCU with a genuine STM32F103C8T6 chip purchased through a reputable supplier.

MeteorSlime, 2024-03-24:
> I actually think they're all fake. Going down that avenue I would expect to replace the fake MCU with a genuine STM32F103C8T6 chip purchased through a reputable supplier.

Where do you Purchase the Blue Pills for your Projects or do you desolder the STM32F103C8T6 every time and replace with genuine one.

keirf, 2024-03-24:
I don't use Blue Pills at all any more, and I haven't done for years. These days I'd generally make a custom PCB and install an Artery (AT32F4xx) chip of some kind.

MeteorSlime, 2024-03-27:
Will my fake Blue Pill work to build a Greaseweazle

keirf, 2024-03-27:
Try it and see. Quite possibly.

MeteorSlime, 2024-03-28:
I tried it and when i typed in the command: ./gw.exe read test.scp --drive=b
It hung for a while and then said this:
** FATAL ERROR:
Write timeout
the drive did not spin during the hanging period.
I don't know if this is because my floppy drive is broken i soldered the pcb incorrectly or because the blue pill is fake.

keirf, 2024-03-28:
Try `./gw.exe --bt read test.scp --drive=b` as this will give a backtrace of the error location. It's probably in the serial comms library though, which isn't promising.

MeteorSlime, 2024-03-28:
It hangs extremly long when i unplug the USB cable while it's hanging then it give me the following error:
Traceback (most recent call last):
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python311\site-packages\cx_Freeze\initscripts\__startup__.py", line 124, in run
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python311\site-packages\cx_Freeze\initscripts\console.py", line 16, in run
  File "gw.py", line 3, in <module>
  File "C:\hostedtoolcache\windows\Python\3.11.8\x64\Lib\site-packages\greaseweazle\cli.py", line 69, in main
  File "C:\hostedtoolcache\windows\Python\3.11.8\x64\Lib\site-packages\greaseweazle\tools\read.py", line 225, in main
  File "C:\hostedtoolcache\windows\Python\3.11.8\x64\Lib\site-packages\greaseweazle\tools\util.py", line 457, in usb_open
  File "C:\hostedtoolcache\windows\Python\3.11.8\x64\Lib\site-packages\greaseweazle\usb.py", line 196, in __init__
  File "C:\hostedtoolcache\windows\Python\3.11.8\x64\Lib\site-packages\greaseweazle\usb.py", line 239, in _send_cmd
  File "C:\hostedtoolcache\windows\Python\3.11.8\x64\Lib\site-packages\serial\serialwin32.py", line 325, in write
serial.serialutil.SerialTimeoutException: Write timeout

keirf, 2024-03-28:
Yes this means usb is not working properly on your fake chip.

keirf, 2024-03-28:
That's only needed if the pill isn't identified as a usb device at all.

MeteorSlime, 2024-03-28:
Well i guess i'll buy a new Blue Pill.

MeteorSlime, 2024-04-17:
My new Blue Pill arrived now i get this error:
Writing c=0-79:h=0-1
Format amiga.amigados
T0.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T0.0: Writing Track (Verify Failure: Retry #1)
T0.0: Writing Track (Verify Failure: Retry #2)
T0.0: Writing Track (Verify Failure: Retry #3)
** FATAL ERROR:
Failed to verify Track 0.0

when trying to write adf images.
I use the following command: ./gw.exe write test.adf --drive=a
Reading and writing scp images works just fine.
I converted my adf image to a scp image and it wrote just fine i still have to test it on my amiga.
I hooked up a PC floppy instead of a real amiga floppy drive.
Is that maybe the problem?

MeteorSlime, 2024-04-17:
I tried it on my amiga and it worked!