keirf/greaseweazle #424: NoIndex

Macgician, 2024-04-11:
I bought three Greaseweazle. One is running, two are not. Im using the same drives, same cables, same disks, same commands. And still two Greasewazle are not working whilst one is doing perfectly fine. Any suggestions?

Macgician, 2024-04-11:
oh I also tried an extern PSU for the drives.  Im using an Alps DF354 and a Mitsumi. Both are running at my working Greaseweazle but not at the others...

keirf, 2024-04-12:
What sort of commands are you trying? What sort of Greaseweazles? Where did you get them?

Macgician, 2024-04-12:
For example gw update works with all three. 

Gw rpm (and all other that are trying to use the floppy) only works with one Greaseweazle (but with all floppies)

They are all Greaseweazle 4

And I bought them at eBay. One never worked the other one used to work and then suddenly stopped working (without changing anything)

I can use gw update on all of them. The two that doesn't work have 1.5 the working one has 1.3. I tried to downgrade one to 1.3 again. Whilst I can still flash a new fw it still won't work

keirf, 2024-04-12:
Well, since you can literally do a side by side swap-out comparison, I'd say the non working ones have hardware faults. Unusual to find two not working though, they're quite reliable generally.

keirf, 2024-04-12:
If you're in the UK you could send them to me to take a look at. Otherwise contact your seller if you didn't buy them so long ago

keirf, 2024-04-12:
In fact you could send them to me regardless but I'm UK based

Macgician, 2024-04-12:
"They are quite reliable" that's actually what I thought, too. Also because they can still be flashed as I assume if anything would take damage then the AT32s.

keirf, 2024-04-12:
That's assuming they are originally from me which they probably are but link to your reseller so I can know for sure.

Macgician, 2024-04-12:
> In fact you could send them to me regardless but I'm UK based

I'm from Germany. I bought them some time  ago and never complained as I thought I'm just using wrong commands. I missed the two week time to send it back and then bought another one which then worked and stopped working. This is a year ago now 🙈

Macgician, 2024-04-12:
> regardless

I'm gonna check the shipment costs. I guess there will not be any customs duty?

keirf, 2024-04-12:
Email keir.xen@gmail.com to arrange return. Customs should not be a problem EU to UK at least, vice versa is more pot luck. Please also let me know by email which eBay seller you purchased from.

keirf, 2024-04-16:
Hi there, Any update? To be clear -- I will repair/replace the units you send me, and cover return shipping cost, if the units are purchased via one of my resellers.

keirf, 2024-04-16:
Sorry I did not receive your email. Check spelling?

Macgician, 2024-04-16:
Hm, I just tapped on your email address and wrote my mail. Have you checked SPAM?

keirf, 2024-04-16:
Yeah I had a good look round my mailbox. Nothing. Weird huh. No typo my end either, this is copy-pasted from my mailbox: keir.xen@gmail.com

Maybe it will arrive later haha.

Macgician, 2024-04-16:
Just resent it

keirf, 2024-04-16:
I received the re-sent version. I will reply shortly, after work meetings :(

closed 2024-04-16 by keirf