keirf/greaseweazle #429: Add syzygy diskdefs for TRS-80 M1

SidTheMole, 2024-04-26:
syzygy did some work last year developing appropriate diskdefs for TRS-80 model 1 disks. As it hasn't yet been added to GW diskdefs releases, it could be.

https://forum.vcfed.org/index.php?threads/greaseweazle-v4-reading-m1-5-25-floppies-on-a-sa400a-drive.1242918/

diskdefs uploaded July 11 2023

disk TRS80SSSD.35T
    cyls = 35
    heads = 1
    tracks * ibm.fm
        secs = 10
	id=0
        bps = 256
        gap1=20
	gap2=20
	gap3=20
        rate = 125
    end
end

disk TRS80SSSD.40T
    cyls = 40
    heads = 1
    tracks * ibm.fm
        secs = 10
	id=1
        bps = 256
        gap1=20
	gap2=20
	gap3=20
        rate = 125
    end
end

disk TRS80NDSSSD.40T
    cyls = 40
    heads = 1
    tracks * ibm.fm
        secs = 10
	id=0
        bps = 256
        gap1=20
	gap2=20
	gap3=20
        rate = 125
    end
end

disk TRS80NDSSDD.40T
    cyls = 40
    heads = 1
    tracks * ibm.mfm
        secs = 18
	id=0
        bps = 256
        gap1=20
	gap2=20
	gap3=40
        rate = 250
    end
end

disk TRS80NDSSDD.35T
    cyls = 35
    heads = 1
    tracks * ibm.mfm
        secs = 18
	id=1
        bps = 256
        gap1=20
	gap2=20
	gap3=40
        rate = 250
    end
end

SidTheMole, 2024-05-19:
In the above thread, a command line  addition was given for TRS-80 LDOS disks - --dd L  .  What does this do?

keirf, 2024-05-19:
`--dd L` pulls pin 2 low. This pin is often used by 5.25-inch drives to determine track density. Default may be HD, and pulling low may be DD (or vice versa!)

keirf, 2024-05-31:
The forum post seems to indicate that these defs are WIP. For example the NewDOS formats are unproven. I will wait and see if syzygy has any more to say on them.

behines, 2024-06-03:
Hi, I've been participating in a discussion about diskdefs for TRS-80 on the Vintage Computers forum, and it was suggested that I post my findings here.  

These findings are for a Model III TRS-80 but I believe apply to Model I as well.

First, for the TRS80SSDD.40T and TRS80DSDD.40T, in addition to the diskdefs above, I believe one should add:

`  interleave = 6`

The disks appear to work with the value set to 1, but the TRSDOS 1.3 disk images I've seen - and have successfully tested on a real Model 3 - have interleave 6.

Then I also did some work with LDOS.  Its sector numbering starts at 0, with an interleave of 3.  An LDOS disk image written with these parameters boots on a real Model III:
```
disk LDOS_SSDD.40T
    cyls = 40
    heads = 1
    tracks * ibm.mfm
        secs = 18
        id=0
        interleave=3
        bps = 256
        gap1=20
	gap2=20
	gap3=40
        rate = 250
    end
end

disk LDOS_DSDD.40T
    cyls = 40
    heads = 2
    tracks * ibm.mfm
        secs = 18
        id=0
        interleave=3
        bps = 256
        gap1=20
        gap2=20
        gap3=40
        rate = 250
    end
end
```
Hope this is useful,
Brad

DrRGee, 2024-06-03:
> The forum post seems to indicate that these defs are WIP. For example the NewDOS formats are unproven. I will wait and see if syzygy has any more to say on them.

Concur completely. They were (and still are) VERY much a Work-in-Progress (WIP). I confess, I have not returned to the project yet. 

As I remember (correct me if in error), NEWDOS [2.0?] allowed you to easily supply Tracks and sectors using the PDRIVE command. It would seem to me that this might make it difficult to have any kind of standard definition.

behines, 2024-06-03:
I want to concur with the concurrence.  I am a relative newbie on this floppy stuff and have just been drinking from the firehose as I've gotten my own system going.  I present my findings as just that, findings for consideration by the community.

Also, I was thinking overnight and it occurs to me that my findings may not apply to Model I at all because I think it only does single density?  So maybe my results need to be sequestered into a "Model 3/Model 4" section.