keirf/greaseweazle #430: Redefinition of _FORTIFY_SOURCE

VorpalBlade, 2024-04-28:
On recent Arch Linux, packages are by default built with:

```
CFLAGS="-march=x86-64-v3 -mtune=generic -O2 -pipe -fno-plt -fexceptions \
        -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security \
        -fstack-clash-protection -fcf-protection \
        -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer"
```

This causes the following error when building greaseweazle:

```
* Building wheel...
running bdist_wheel
running build
running build_py
copying src/greaseweazle/__init__.py -> build/lib.linux-x86_64-cpython-312/greaseweazle
running build_ext
building 'greaseweazle.optimised.optimised' extension
gcc -fno-strict-overflow -DNDEBUG -g -O3 -Wall -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -ffat-lto-objects -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -g -ffile-prefix-map=/build/python/src=/usr/src/debug/python -flto=auto -march=x86-64-v3 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -flto=auto -fPIC -I/usr/include/python3.12 -c src/greaseweazle/optimised/c64.c -o build/temp.linux-x86_64-cpython-312/src/greaseweazle/optimised/c64.o -Wall -Werror
<command-line>: error: "_FORTIFY_SOURCE" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
cc1: all warnings being treated as errors
error: command '/usr/bin/gcc' failed with exit code 1
```

This is because you use the following in your setup.py:

```
extra_compile_args = ['-Wall', '-Werror']
```

I'm not sure what the best way forward is. I'm also not sure where the additional `-D_FORTIFY_SOURCE=2` comes from, maybe it is something from setuptools itself? I could of course patch greazeweasle in my Arch Linux package to remove the `-Werror` but that isn't great either,

keirf, 2024-04-28:
I think patching the `-Werror` is okay. It shouldn't affect the generated code after all. Maybe longer term we can work out something neater.

keirf, 2024-04-29:
Closing this ticket. External patching is an appropriate solution here, imo.

closed 2024-04-29 by keirf