keirf/greaseweazle #431: error: externally-managed-environment

Burrito78, 2024-04-29:
Hi,

when trying to update to the latest version using
`python3 -m pip install git+https://github.com/keirf/greaseweazle@latest`

I'm getting the following error
```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try brew install
    xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a Python library that isn't in Homebrew,
    use a virtual environment:
    
    python3 -m venv path/to/venv
    source path/to/venv/bin/activate
    python3 -m pip install xyz
    
    If you wish to install a Python application that isn't in Homebrew,
    it may be easiest to use 'pipx install xyz', which will manage a
    virtual environment for you. You can install pipx with
    
    brew install pipx
    
    You may restore the old behavior of pip by passing
    the '--break-system-packages' flag to pip, or by adding
    'break-system-packages = true' to your pip.conf file. The latter
    will permanently disable this error.
    
    If you disable this error, we STRONGLY recommend that you additionally
    pass the '--user' flag to pip, or set 'user = true' in your pip.conf
    file. Failure to do this can result in a broken Homebrew installation.
    
    Read more about this behavior here: <https://peps.python.org/pep-0668/>

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
```

I re-installing Python using `brew install python` doesn't help.

keirf, 2024-04-29:
Yes, many distros are moving to require using a virtual environment for external packages. You can basically use the instructions in the error message, or the virtual environment instructions in the wiki (Linux specific at the moment, but I will extend them for macos).

keirf, 2024-04-29:
Closing but please reopen if you see any further issues regarding the updated instructions for using a virtual environment.

closed 2024-04-29 by keirf

Burrito78, 2024-04-29:
OK, it is working now. Thanks for the fast and kind help as always!

But it sure is a bummer that I have to remember this activate command before using the GW software.

keirf, 2024-04-29:
You don't *have* to use the activate command I believe -- you could put the path/to/bin in your PATH variable, or directly specify the path when invoking gw for example `greaseweazle/bin/gw`

keirf, 2024-04-29:
Another alternative is a symlink from a location that is on your default path. I believe default paths are quite locked down in macOS however. I have added some more text to the wiki but, on balance, the easiest thing to do is just remember to `activate`.

keirf, 2024-04-29:
Another alternative is to use `pipx` which manages all the venv stuff automatically and probably puts `gw` on your default path. I will investigate that and possibly recommend that instead.

keirf, 2024-04-30:
Okay, `pipx` looks pretty neat. I updated the instructions to recommend this new method. I think it is more along the lines of what you'd expect as an experience.