keirf/greaseweazle #432: OS9 Configuration

UFO-OS9, 2024-05-02:
Hello,
I have been trying to create an image of an OS9 floppy for use on a Gotek with Flash floppy.
the Greaseweazle creates an image without errors & writes  a floppy without errors, but the diskette & the image do not work in the OS9 unit. I assume this is due to my settings. ?

The working Gotek settings:
Microware OS-9 system requires jumper at S0 and the following lines in FF.CFG. Emulation will work with 1.6MB IMG files.
interface = ibmpc
host = akai
Hence I have been setting the GW  format to Akai 1600 & 800 & others to no avail !

The manufacturers 1.6MB IMG file write parameters on their CMD line script, that writes OS9 images to floppy via a windows PC are:
diskrw A: W -t=80 -s=10 -b=1024 -f=E5 -n ... if this helps

But I cannot find a combination in GW that works.

Any help would be most appreciated, thanks for listening !

keirf, 2024-05-02:
What command line are you using to generate the image using Greaseweazle? Can you zip the image file, and also a raw flux image file (eg. `gw read my.scp`), and attach to this ticket. Also include your FF.CFG file in the zip archive.

UFO-OS9, 2024-05-02:
Hi Keirf,
I just reposted this in the discussions thinking this was the wrong place.. sorry.
I had no joy with the terminal , so I've been using the GUI which generates:
 gw.exe read --format=akai.1600 --device=COM5 "C:\Users\User\Documents\GD-Service\Grease Weasel\Images\1.img"

Requested files attached:
[Image-files.zip](https://github.com/keirf/greaseweazle/files/15186508/Image-files.zip)

Thanks for your help !

keirf, 2024-05-02:
The scp file isn't really raw because `--format akai.1600` is still on the command line. You need to get rid of that. Maybe there's a "no format" option somewhere in the GUI?

keirf, 2024-05-02:
Another thing to consider: Is the Microware system and its cabling known good? Eg. Can you cross-check with a real floppy drive in the Microware system?

keirf, 2024-05-02:
I believe shift + right click in the greaseweazle folder, then select "Open command window here". Then that will open a command window where you can run gw.exe.

closed 2024-05-02 by keirf