keirf/greaseweazle #435: 5.25 loppy drives with density pin

briskspirit, 2024-05-15:
Hi all!

I have a Compaq Portable 3 floppy drive that supports both double density and high density 5.25 floppy diskettes. 
When it is connected to original PC - it works fine with both densities and shows it with LED being orange for DD and green for HD.

I have tried to use gw pi set 2 high (or low) to toggle density selection and floppy reacts with the proper LED color but on read or write gw still tries to step as it is HD diskette (with 80 track iirc.) That's considering that I explicitly select type as ibm 360. 

How to fix this issue? Do I have to make manual script where I should skip every even track? And why GW assumes 80 tracks per disk if ibm 360 was selected?

Also thanks a lot for such a great product! Works perfectly with my 8" Shugart.

keirf, 2024-05-16:
GW "becomes" a 40 cyl drive when a 40 cyl image is mounted -- one step per cylinder.

Is the Compaq double-stepping for 48tpi disk types?

keirf, 2024-05-31:
A more coherent response: Try `--tracks=step=2` on your command line, to enable double-stepping for 48tpi disks.

Let me know how you get on.

briskspirit, 2024-06-24:
> GW "becomes" a 40 cyl drive when a 40 cyl image is mounted -- one step per cylinder.
> 
> Is the Compaq double-stepping for 48tpi disk types?

I understand how it can "become" 40 cyl when writing, but I had issue with reading.

> A more coherent response: Try `--tracks=step=2` on your command line, to enable double-stepping for 48tpi disks.
> 
> Let me know how you get on.

I don't have a chance to check it with Compaq drive anymore, but got Canon MD5511 (just a dual drive). It also had same "issue" as in my initial post(reading SD floppy in DD drive) but your fix helped! Think would be nice to have optional key like "--drive_density" for that?

briskspirit, 2024-06-24:
Also seems I need this option to double step if I want to write SD image (ibm.360) with DD floppy drive, but write function doesn't have it.

keirf, 2024-06-24:
You can specify --step=2 to gw write as well. However you cannot reliably write a 48tpi disk with a 96tpi drive because it will write tracks which are too narrow.