keirf/greaseweazle #441: greaseweazle-1.17-win64.zip has a virus

cryton2707, 2024-05-31:
Bitdefender is Identifying this file as being virusd

We blocked this dangerous page for your protection:

https://objects.githubusercontent.com/github-production-release-asset-2e65be/126182850/63dbae7b-f805-48f4-9096-2f7c5cfde52f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20240531%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240531T224210Z&X-Amz-Expires=300&X-Amz-Signature=2fa1ab23a028bb3cc8538fa143f004a263503d49e06ec26377cafe127550094d&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=126182850&response-content-disposition=attachment%3B%20filename%3Dgreaseweazle-1.17-win64.zip&response-content-type=application%2Foctet-stream

Threat name: **Trojan.GenericKD.72928753**
Dangerous pages attempt to install software that can harm the device, gather personal information or operate without your consent.

cryton2707, 2024-05-31:
32bit version appears ok.

keirf, 2024-06-01:
It will be a false positive. I've seen these from time to time with Windows Defender, which of course is the most common report as nearly everyone runs that. Unfortunately it's a perennial problem with unsigned software, and especially it seems with packaged Python applications. Bypass it if you can and you feel that's okay; otherwise running a slightly older version, or the win32 version, is the only way (as you have done).

closed 2024-06-01 by keirf