keirf/greaseweazle #442: Unable to use --adjust-speed parameter with Apple II formats

Digitoxin1, 2024-06-02:
The following command results in an error:

`gw read --drive 2 --format apple2.appledos.140 --adjust-speed 300rpm --tracks="step=2" --raw flux\disk1\track00.0.raw`

I get the following error:

`usage: gw.exe read [options] file
gw.exe read: error: the following arguments are required: file`

The command works properly if I remove the --adjust-speed 300rpm parameter.  This only appears to be an issue when using --format apple2.appledos.140.

keirf, 2024-06-03:
Although I hate to close this without a fix, I can't actually reproduce this (albeit on Linux). Could you add a `--bt` option and collect a backtrace if you can still make this failure happen. Like this: `gw --bt read --drive 2 [...]`

Digitoxin1, 2024-06-03:
Nevermind.  This was my error.  I was using a batch file and went over the available nine parameters.

closed 2024-06-03 by Digitoxin1