keirf/greaseweazle #444: Apple II raw images created with Greaseweazle do not read properly in other applications

Digitoxin1, 2024-06-02:
DTC.exe and HxCFloppyEmulator are unable to properly read and convert flux images of Apple II disks created with Greaseweazle 1.17.

keirf, 2024-06-03:
So to be clear, you write a disk and it works in an Apple II but it is unrecognised by these tools if you image it?

keirf, 2024-06-03:
Does Greaseweazle handle the dumped flux image okay? Can you share one? I did test generated HFE and SCP image files in HxC and they worked okay.

Digitoxin1, 2024-06-03:
I have an unprotected AppleDOS Apple II disk.

Using the following command, I make a raw flux set from the disk

`gw read --drive 2 --format apple2.appledos.140 --tracks="step=2" --raw flux\disk1\track00.0.raw`

I then run the following command to make a .do file

`gw convert --format apple2.appledos.140 flux\disk1\track00.0.raw disk1.do`

This generates a working disk image that boots perfectly in the AppleWin emulator.

If I try to use HxCFloppyEmulator or DTC.exe to convert that same flux set into a .do file, I end up with an unusable disk image.

keirf, 2024-06-03:
Ohhh I see. It's an original Apple2 disk but the flux file created by Greaseweazle doesn't work with other tools. I see. I'll have a look.

keirf, 2024-06-03:
The dumps generated by your drive contain lots of false flux transitions. This is because your drive is not expecting to read such long flux samples (up to 12 microseconds). You may get a better dump by specifying `--densel L` to `gw read`. However, Greaseweazle is able to discard the false transitions, hence it can fully decode the dump.

In short: Not a bug, actually a feature (of Greaseweazle).

closed 2024-06-03 by keirf