keirf/greaseweazle #445: Dead/Bad Greaseweazle? Pins 10,34 shorted to ground. Unable to read disks.

anetanel, 2024-06-07:
I got my new Greaseweazle v4.1 from retropassion.co.uk, and after couple of uses, it stopped working reliably.
When reading flux I'm getting what seems to just noise, and when reading a simple .img, I'm getting "Ignoring unexpected sector" for each sector.
I suspected the usb cable, the floppy cable, and the floppy drive, so I replaced all of them but to no avail.
I also have a Kryoflux which works fine in the same setup.
I noticed that when connecting the floppy ribbon cable to the Greaseweazle, the motor starts spinning, even when the Greaseweazle is disconnected from the computer.
I tested the Greaseweazle 34 pins with a multimeter, a found that pins 10 and 34 are shorted to ground. I compared it with the Kryoflux and there the pins are not shorted.
Looking at the schematics (with my limited electronics knowledge...) it seems that these pins shouldn't be shorted...
Is my Greaseweazle toast?

keirf, 2024-06-07:
Are you certain that you are connecting the ribbon the correct way round? Bear in mind that it is "upside down" at the Greaseweazle end compared with a typical floppy drive. It is very easy to get this muddled and overlook this if your ribbon connector isn't keyed.

Basically, try flipping the connector at the Greaseweazle end, if you can. If this doesn't help then your Greaseweazle is sadly buggered. I assume you are in the UK and I'll be happy to send you a replacement directly in this case. Just email me (address is in my GitHub profile) with your shipping address. I supply Retropassion so perfectly happy to sort out warranty issues directly!

anetanel, 2024-06-07:
I'm sure that the cable is oriented correctly. It is keyed, and the colored wire is on the right, aligned with the pin 1 "arrow" mark on the board.
I assume that the motor is spinning because pin 10 is shorted on the board?
I'm from Israel... :\

keirf, 2024-06-07:
Well that sounds about right. Of course this puts pin 10 on the bottom row, and all the top row are ground pins. That matches your testing? Another thing you can try is a *straight* ribbon (no twist across pins 10-16) and identify as `--drive=b` in greaseweazle tools. This would use pin 12 for select and pin 16 for motor, conveniently avoiding your suspicious pin 10.

Well, shipping to Israel is not actually a problem. I could certainly send you one by standard untracked airmail. I suppose it would take a couple of weeks?

anetanel, 2024-06-07:
Using the straight part of the cable did "solve" the spinning motor issue, and I'm able to seek and get rpm reading on drive "b".
But reading disks still results in an "unexpected sector" error for all tracks:
```
$ gw read --format ibm.360 --drive b testgw.img
Reading c=0-39:h=0-1 revs=2
Format ibm.360
T0.0: Ignoring unexpected sector C:0 H:1 R:7 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:2 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:9 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:3 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:5 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:8 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:6 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:1 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:4 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:7 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:2 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:9 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:3 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:5 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:8 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:6 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:1 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:4 N:2
T0.0: IBM MFM (0/9 sectors) from Raw Flux (85333 flux in 401.71ms)
.
.
T0.0: Ignoring unexpected sector C:0 H:1 R:1 N:2
T0.0: Ignoring unexpected sector C:0 H:1 R:4 N:2
T0.0: IBM MFM (0/9 sectors) from Raw Flux (469196 flux in 2209.53ms) (Retry #1.3)
T0.0: Giving up: 9 sectors missing
T0.1: IBM MFM (9/9 sectors) from Raw Flux (85334 flux in 401.90ms)
T1.0: Ignoring unexpected sector C:1 H:1 R:9 N:2
T1.0: Ignoring unexpected sector C:1 H:1 R:4 N:2
T1.0: Ignoring unexpected sector C:1 H:1 R:5 N:2
.
.
T1.0: Ignoring unexpected sector C:1 H:1 R:3 N:2
T1.0: Ignoring unexpected sector C:1 H:1 R:6 N:2
T1.0: IBM MFM (0/9 sectors) from Raw Flux (194986 flux in 1003.63ms) (Retry #1.1)
``` 
etc...

while using the exact same setup with KryoFlux results in a perfectly good image:
```
$ dtc -ftest.img -i4 -v300 -n-9 -e40 -d1
KryoFlux DiskTool Console, v3.00_OSX_64bit, uiv.1, Jun 23 2021, 15:45:43
(c) 2009-2021 KryoFlux Products & Services Ltd.
Developed by The Software Preservation Society, www.softpres.org
Licensed for private, non-commercial use only.

00.0    : frev: 44726, drift: 0.125 us, tfer: 208636 B/s, rpm: 298.995
00.0    : base: 1.999 us [99.857%], band: 4.001 us, 5.996 us, 8.020 us
00.0    : MFM: OK*, trk: 000, sec: 9, *H +3
00.1    : frev: 42620, drift: 0.062 us, tfer: 197325 B/s, rpm: 298.970
00.1    : base: 1.999 us [99.907%], band: 3.996 us, 5.996 us, 8.008 us
00.1    : MFM: OK*, trk: 000, sec: 9, *H +2
01.0    : frev: 38558, drift: 0.021 us, tfer: 179978 B/s, rpm: 299.047
01.0    : base: 2.000 us [99.928%], band: 4.000 us, 6.000 us, 8.012 us
01.0    : MFM: OK*, trk: 001, sec: 9, *H +5
01.1    : frev: 38960, drift: 0.229 us, tfer: 177539 B/s, rpm: 299.367
01.1    : base: 2.000 us [99.947%], band: 3.996 us, 6.001 us, 8.005 us
01.1    : MFM: OK*, trk: 001, sec: 9, *H +4
02.0    : frev: 38819, drift: 0.728 us, tfer: 181641 B/s, rpm: 299.070
02.0    : base: 2.000 us [99.867%], band: 3.998 us, 6.001 us, 8.021 us
02.0    : MFM: OK*, trk: 002, sec: 9, *H +3
02.1    : frev: 38954, drift: 0.895 us, tfer: 177539 B/s, rpm: 299.088
02.1    : base: 2.001 us [99.949%], band: 3.995 us, 6.003 us, 7.999 us
02.1    : MFM: OK*, trk: 002, sec: 9, *H +3
.
.
.
39.1    : frev: 38422, drift: 0.104 us, tfer: 175635 B/s, rpm: 298.736
39.1    : base: 2.010 us [99.227%], band: 3.911 us, 6.031 us, 8.019 us
39.1    : MFM: OK*, trk: 039, sec: 9, *H +1
40.0    : frev: 25236, drift: 1.290 us, tfer: 117615 B/s, rpm: 299.259
40.0    : base: 2.000 us [66.395%], band: 3.968 us, 8.000 us
40.0    : MFM: <unformatted>
40.1    : frev: 25236, drift: 0.166 us, tfer: 113736 B/s, rpm: 299.295
40.1    : base: 2.001 us [66.306%], band: 3.959 us, 8.003 us
40.1    : MFM: <unformatted>

Enjoy your shiny new disk image!
Please consider helping us to preserve media and continue development:
www.softpres.org/donate

$ file test.img
test.img: DOS/MBR boot sector, code offset 0x34+2, OEM-ID "TAN  3.3", sectors/cluster 2, root entries 112, sectors 720 (volumes <=32 MB), Media descriptor 0xfd, sectors/FAT 2, sectors/track 9, dos < 4.0 BootSector (0), FAT (12 bit by descriptor), followed by FAT
````

So I guess whatever is wrong with the Greaseweazle is more than just a "bad" pin.

keirf, 2024-06-07:
Thanks, this probably indicates that the side-select pin is also pulling low. Nonetheless could you do a raw Kryoflux dump (to stream files), using your Kryoflux, of at least a few cylinders, zip them and attach them here? Then we can trivially check the sector headers on sides 0 and 1 and confirm this hypoothesis.

Another thing to check is that your drive has pull-up terminations correctly installed/selected. Greaseweazle outputs are open collector and require these terminations. Kryoflux has push-pull outputs, doesn't require terminations.  This is a bit of a long shot since (a) some pins are obviously working! (b) your greaseweazle used to work.

anetanel, 2024-06-07:
Indeed, pin 32 is also shorted. I guess I missed that because it is marginal (almost 10 ohm) and my too fancy for its own good multimeter takes some time to register it as short.
Here is the raw flux data from Kruoflux.
[test_kf.zip](https://github.com/user-attachments/files/15738422/test_kf.zip)

I also find it hard to believe this is a termination issue, but I tested it anyway.
I'm using a Tandon TM-100 and I took the resistor block (the blue chip) off, but It did not change much. Still getting "Ignoring unexpected sector" with the Greaseweazle.
[image]

closed 2024-06-07 by keirf