keirf/greaseweazle #446: Commodore 1541 format: 40 vs 35 tracks

nroach44, 2024-06-10:
All the online sources I've found indicate that the 1541 (at least as used by a C64) only used the first 35 tracks of the disk.

When dumping my own disks, they all seem to "fail" on tracks 36-40:

```
Cyl-> 0         1         2         3         
H. S: 0123456789012345678901234567890123456789
0. 0: ...................................XXXXX
0. 1: ...................................XXXXX
0. 2: ...................................XXXXX
0. 3: ...................................XXXXX
0. 4: ...................................XXXXX
0. 5: ...................................XXXXX
0. 6: ...................................XXXXX
0. 7: ...................................XXXXX
0. 8: ...................................XXXXX
0. 9: ...................................XXXXX
0.10: ...................................XXXXX
0.11: ...................................XXXXX
0.12: ...................................XXXXX
0.13: .........X.........................XXXXX
0.14: ...................................XXXXX
0.15: ...................................XXXXX
0.16: ...................................XXXXX
0.17: ..............................          
0.18: ........................                
0.19: .................                       
0.20: .................                       
Found 682 sectors of 768 (88%)
```

Is there a reason gw attempts to read these extra tracks by default? I've not used the C64 myself, I'm just dumping some old family disks.

keirf, 2024-06-10:
C64 isn't my system either, but by all accounts it wasn't that uncommon to squeeze in extra tracks. Hence it makes sense to try dumping them because otherwise you would miss those tracks and make bad images. If you are confident that your disks are all going to be the regular 35-track sort, you can add to your command line: `--tracks=c=0-34` will reduce the cylinder range.

keirf, 2024-06-10:
Arguably the default should be 0-34 and need override for more, but the risk of missing tracks by default got to me.

nroach44, 2024-06-10:
No worries! I thought it might have been that and wanted to check. 

I'd vote for setting 35 the default since it's more "correct" but I've got no idea how widespread 40-track disks are.

keirf, 2024-06-10:
Actually given how I've hacked in the 35- to 40-track variability, it's probably not entirely trivial to change it. So I won't. :)

closed 2024-06-10 by keirf