keirf/greaseweazle #447: Best practice for reading 800k Mac floppy disks and convert to disk image files?

wilhelmloof, 2024-06-13:
### What is best practice for reading 800k Mac floppy disks and converting them to disk image files?

I've been reading the disks with this command:

    gw read disk-00.0.raw

And converting the raw flux data to a binary disk image file like this:

    gw convert --format mac.800 disk-00.0.raw disk.img

Or reading directly to a disk image: 

    gw read --format mac.800 disk.img

There are several other switches/arguments available under `gw read --help`, and I wonder if there are some other arguments that can be useful for reading or converting, in order to make an img-file as good as possible?

Is there any benefit adding `--format mac.800` also when reading raw data?

What's a recommended method for exporting all files in an img-file on macOS/bash?

keirf, 2024-06-14:
If you add `--format mac.800` when reading raw, then by default the "raw" image will actually contain re-created tracks after being passed through the Greaseweazle Mac GCR codec. You probably don't want that! However if you also add `--raw` then you will get the benefit of track verification against the `mac.800` format while still writing out the raw flux direct from disk.

keirf, 2024-06-14:
Beyond that, not really much to say. I try to design the command line to work well without needing lots of options.

closed 2024-06-14 by keirf

wilhelmloof, 2024-06-14:
Thanks for clarifying.

So, if I understand you correctly, the command for track verification while still writing raw flux data would be:

    gw read --format mac.800 --raw disk-00.0.raw

And the written flux data with the command above will be the same the plain command:

    gw read disk-00.0.raw

When is verification useful? Do you often read disks with verification? What happens if there are format errors while verifying, will gw still write raw flux data?