keirf/greaseweazle #454: IBM's 2.4meg 5 1/4" floppy format craziness

WarlockD, 2024-07-05:
Not sure if this is a REAL issue or not, but I picked up one of those 2.4meg floppy drives  that IBM use on some of their controllers cheap and thought it be perfect for scanning disks.  It even had  a 2.4meg floppy in it!

I thought I would just scan the disk as a normal 1.2meg at the start then try running it at 1mbps.  Come to find out I found something weird.  The first cylinder of both heads came up as standard 15 sectors as a 1.2meg disk with no crc errors, but the rest was all garbage.  What was wierd was it didn't look like any kind of file format, it looked more like magnetic tape file.

After some digging I found a note about pin 2 on how it lowers the speed of the drive to 180rpm.  I read the drive again and lo and behold I got the entire disk image (all 128meg scp of it) and had clearly 30 tracks formatted from an old fashioned 765 chip with no CRC erros.  However, the first two tracks come up with tons of garbage and CRC errors even with three copy's.

So..er how do I save this disk?  I am not too concerned as this is just a firmware disk, but I was wondering why it read the data on the first cylinder fine at 360rpm but at 180rpm, it read a lot of errors.  I would of thought the data would of been the same or is it something to do with the drive hardware itself?

PS - Its been kind of fun custom formatting an IBM disk at 2.4 meg.  Seems to work fine on some of the other 1.2meg disks I have.  Might dig into the drive to see if its just a simple current limiter or if you can do something to adjust the speed.

keirf, 2024-07-05:
The drive electronics will only correctly read data at the expected rate. So you need to change pin 2 during dump. That's not currently possible but what you can do is two dumps, and dump to Kryoflux stream files which are file per track anyway. Then you could finally convert those to scp if you want to.

keirf, 2024-07-05:
For example something like:
```
gw read --tracks=c=0 --dd=H mydump/00.0.raw
gw read --tracks=c=1-79 --dd=L mydump/00.0.raw
gw convert mydump/00.0.raw mydump.scp
```
Later:
```
gw write --tracks=c=0 --dd=H mydump.scp
gw write --tracks=c=1-79 --dd=L mydump.scp
```

The conversion to SCP is optional of course. You can just work with kryoflux stream instead.

Also we could add a diskdef for these high-capacity disks. Then you could convert to IMG.

WarlockD, 2024-07-05:
@keirf Ah that worked!  Honestly am am working on a pull to put in floppy tape support and maybe experimenting on making the pin 2 run a pwn to see if I can get the drive to not only run at 180rpm but 300 as well.

That said it only be for QIC-113 and might be a bit as I have to dig out my old drives

closed 2024-08-11 by keirf