keirf/greaseweazle #457: Conversion of verified IBM 720 raw dump to IMG fails

Litude, 2024-07-16:
I dumped a disk that had some read errors during dumping, but after retries all sectors were dumped successfully according to the logs during the dumping process:
```
> gw read --tracks=c=0-81 --retries 5 --seek-retries 4 --format ibm.720 --revs 5 --raw raw/MBASH4_00.0.raw

Reading c=0-81:h=0-1 revs=5
Format ibm.720
T0.0: IBM MFM (9/9 sectors) from Raw Flux (215994 flux in 1000.40ms)
T0.1: IBM MFM (9/9 sectors) from Raw Flux (215845 flux in 999.81ms)
T1.0: IBM MFM (9/9 sectors) from Raw Flux (189925 flux in 999.77ms)
T1.1: IBM MFM (9/9 sectors) from Raw Flux (190034 flux in 999.56ms)
T2.0: IBM MFM (9/9 sectors) from Raw Flux (190090 flux in 999.67ms)
T2.1: IBM MFM (9/9 sectors) from Raw Flux (190484 flux in 999.51ms)
T3.0: IBM MFM (9/9 sectors) from Raw Flux (190406 flux in 999.65ms)
T3.1: IBM MFM (9/9 sectors) from Raw Flux (190426 flux in 999.51ms)
T4.0: IBM MFM (9/9 sectors) from Raw Flux (190440 flux in 999.65ms)
T4.1: IBM MFM (9/9 sectors) from Raw Flux (190155 flux in 999.50ms)
T5.0: IBM MFM (9/9 sectors) from Raw Flux (189485 flux in 999.58ms)
T5.1: IBM MFM (9/9 sectors) from Raw Flux (189974 flux in 999.47ms)
T6.0: IBM MFM (9/9 sectors) from Raw Flux (189906 flux in 999.56ms)
T6.1: IBM MFM (9/9 sectors) from Raw Flux (189854 flux in 999.48ms)
T7.0: IBM MFM (9/9 sectors) from Raw Flux (189764 flux in 999.57ms)
T7.1: IBM MFM (9/9 sectors) from Raw Flux (190554 flux in 999.45ms)
T8.0: IBM MFM (9/9 sectors) from Raw Flux (189946 flux in 999.56ms)
T8.1: IBM MFM (9/9 sectors) from Raw Flux (190874 flux in 999.46ms)
T9.0: IBM MFM (9/9 sectors) from Raw Flux (190246 flux in 999.55ms)
T9.1: IBM MFM (9/9 sectors) from Raw Flux (190234 flux in 999.45ms)
T10.0: IBM MFM (9/9 sectors) from Raw Flux (189917 flux in 999.50ms)
T10.1: IBM MFM (9/9 sectors) from Raw Flux (190415 flux in 999.45ms)
T11.0: IBM MFM (9/9 sectors) from Raw Flux (190234 flux in 999.49ms)
T11.1: IBM MFM (9/9 sectors) from Raw Flux (190175 flux in 999.43ms)
T12.0: IBM MFM (9/9 sectors) from Raw Flux (189655 flux in 999.48ms)
T12.1: IBM MFM (9/9 sectors) from Raw Flux (189725 flux in 999.45ms)
T13.0: IBM MFM (9/9 sectors) from Raw Flux (189933 flux in 999.49ms)
T13.1: IBM MFM (9/9 sectors) from Raw Flux (189974 flux in 999.44ms)
T14.0: IBM MFM (9/9 sectors) from Raw Flux (189506 flux in 999.50ms)
T14.1: IBM MFM (9/9 sectors) from Raw Flux (189841 flux in 999.42ms)
T15.0: IBM MFM (9/9 sectors) from Raw Flux (190190 flux in 999.47ms)
T15.1: IBM MFM (9/9 sectors) from Raw Flux (190423 flux in 999.45ms)
T16.0: IBM MFM (9/9 sectors) from Raw Flux (189797 flux in 999.48ms)
T16.1: IBM MFM (9/9 sectors) from Raw Flux (189864 flux in 999.44ms)
T17.0: IBM MFM (9/9 sectors) from Raw Flux (189464 flux in 999.51ms)
T17.1: IBM MFM (9/9 sectors) from Raw Flux (189582 flux in 999.45ms)
T18.0: IBM MFM (9/9 sectors) from Raw Flux (189384 flux in 999.53ms)
T18.1: IBM MFM (9/9 sectors) from Raw Flux (190515 flux in 999.44ms)
T19.0: IBM MFM (9/9 sectors) from Raw Flux (189944 flux in 999.49ms)
T19.1: IBM MFM (9/9 sectors) from Raw Flux (191335 flux in 999.45ms)
T20.0: IBM MFM (9/9 sectors) from Raw Flux (190386 flux in 999.49ms)
T20.1: IBM MFM (9/9 sectors) from Raw Flux (189869 flux in 999.44ms)
T21.0: IBM MFM (9/9 sectors) from Raw Flux (189765 flux in 999.50ms)
T21.1: IBM MFM (9/9 sectors) from Raw Flux (189824 flux in 999.42ms)
T22.0: IBM MFM (9/9 sectors) from Raw Flux (190023 flux in 999.47ms)
T22.1: IBM MFM (9/9 sectors) from Raw Flux (190083 flux in 999.42ms)
T23.0: IBM MFM (9/9 sectors) from Raw Flux (189686 flux in 999.45ms)
T23.1: IBM MFM (9/9 sectors) from Raw Flux (190284 flux in 999.42ms)
T24.0: IBM MFM (9/9 sectors) from Raw Flux (190664 flux in 999.43ms)
T24.1: IBM MFM (9/9 sectors) from Raw Flux (191193 flux in 999.43ms)
T25.0: IBM MFM (9/9 sectors) from Raw Flux (191393 flux in 999.46ms)
T25.1: IBM MFM (9/9 sectors) from Raw Flux (190387 flux in 999.43ms)
T26.0: IBM MFM (9/9 sectors) from Raw Flux (189495 flux in 999.46ms)
T26.1: IBM MFM (9/9 sectors) from Raw Flux (189956 flux in 999.42ms)
T27.0: IBM MFM (9/9 sectors) from Raw Flux (191176 flux in 999.46ms)
T27.1: IBM MFM (9/9 sectors) from Raw Flux (190894 flux in 999.43ms)
T28.0: IBM MFM (9/9 sectors) from Raw Flux (190527 flux in 999.45ms)
T28.1: IBM MFM (9/9 sectors) from Raw Flux (189829 flux in 999.41ms)
T29.0: IBM MFM (9/9 sectors) from Raw Flux (189785 flux in 999.49ms)
T29.1: IBM MFM (9/9 sectors) from Raw Flux (190456 flux in 999.45ms)
T30.0: IBM MFM (9/9 sectors) from Raw Flux (190303 flux in 999.42ms)
T30.1: IBM MFM (9/9 sectors) from Raw Flux (190654 flux in 999.40ms)
T31.0: IBM MFM (9/9 sectors) from Raw Flux (190605 flux in 999.42ms)
T31.1: IBM MFM (9/9 sectors) from Raw Flux (190820 flux in 999.42ms)
[... 451 lines cut ...]
0. 0: ................................................................................
0. 1: ................................................................................
0. 2: ................................................................................
0. 3: ................................................................................
0. 4: ................................................................................
0. 5: .....................................................................X..........
0. 6: ................................................................................
0. 7: ................................................................................
0. 8: ................................................................................
1. 0: ................................................................................
1. 1: ................................................................................
1. 2: ................................................................................
1. 3: ................................................................................
1. 4: ................................................................................
1. 5: ................................................................................
1. 6: ................................................................................
1. 7: ................................................................................
1. 8: ................................................................................
Found 1439 sectors of 1440 (99%)
```

I think my drive might need cleaning but I still expected it to get me a perfect dump since no errors occurred during the dumping.

Here is the raw dump:
[MBASH4_RAW.zip](https://github.com/user-attachments/files/16253302/MBASH4_RAW.zip)

keirf, 2024-07-16:
Wow, that is one marginal dump. It's amazing that all the sectors decode at all! Since this occurs on later tracks, I wonder if the density-selection is incorrect? Is this from a 5.25inch drive? Then you may need to specify `--densel=L` to `gw read`. If from a 3.5-inch drive it is possible that the physical density-select switch isn't working properly.

Anyway, back to why this doesn't decode corrcetly on subsequent `gw convert`. It is because the flux doesn't get converted to bitcells in *exactly* the same way.
1. On read, the subsequent read retries get passed through the PLL (clock recovery) individually, with PLL parameters reset on each retry
2. On convert, all the read retries are concatenated together and passed through PLL in one pass.

Anyway, as it happens the data is recoverable, you just need to play with the PLL parameters a bit. So there's something you probably didn't know about. By default, clock recovery tries two different PLL setups.
1. An aggressive PLL (period=5:phase=650)
2. A conservative PLL (period=1:phase=10)

You don't really need to know too much about what those parameters mean, but in almost all cases it is the *conservative* PLL that is saving your bacon when you have a marginal, noisy dump. So it makes sense to try alternative parameters close by to that default conservative setup. In this specific case, this works for me:
```
gw convert MBASH4_00.0.raw a.img --format=ibm.720 --pll=period=1:phase=15
```

So, all is not lost :)

Litude, 2024-07-16:
Thank you! I had no idea it worked like that, but the command you provided works like a charm.

It is a 3.5 inch drive and the drive has worked properly reading other disks. I think this disk possibly had some dirt or something on it that accumulated on the read head which caused the later sector reads to get very noisy. Have had similar things happen before. I'll have to clean the heads and also try redumping it with another drive since this dump ended up a bit too noisy for my liking.

closed 2024-08-11 by keirf