keirf/greaseweazle #460: Converting back side to D64 not possible

klaussoft, 2024-07-22:
I had read a 1541 disk with a flippy modded drive to a SCP file.
With the greaseweazle tools it is only possible to extract a D64 from side A. For extracting side B it is nessesary to reverse the data. There is no possibility to tell greaseweazle tool to reverse the data to extract the D64 image. With g64conv it is possible, so the read data is ok.

The same is by playing back a D64 to the floppy. There is a need for a switch to reverse the data.

keirf, 2024-07-22:
Thanks for this. I'm away this week but will be back in touch from the weekend to try out some patches.

keirf, 2024-07-24:
Can you please supply a flippy flux file as gathered from Greaseweazle. For example as SCP or Kryoflux tream files? This would allow me to do some testing for this feature via "gw convert" and then you will only need to test the "gw read" and "gw write" changes.

klaussoft, 2024-07-24:
[test_ab_full.zip](https://github.com/user-attachments/files/16366318/test_ab_full.zip)
[test_ab.zip](https://github.com/user-attachments/files/16366321/test_ab.zip)
I uploaded two files. Both are from the same 1541 disk.
test_ab is read with step 2 (Tracks 0-39)
test_ab_full is read with 80 tracks (with halftracks)

markusC64, 2024-08-15:
From my set of test images - those are my flippy disk test images for 1541 (and compatible):

[sampleimages-flippydisk.7z.zip](https://github.com/user-attachments/files/16624542/sampleimages-flippydisk.7z.zip)

keirf, 2024-08-21:
I have a new `gw convert` option for you to try. This is on branch `flippy-reverse`. You can find Windows build here (you need to be logged in to GitHub to see the Artifact links):
https://github.com/keirf/greaseweazle/actions/runs/10487456321

Or if on Mac or Linux you can try installing with pipx:
```
pipx install git+https://github.com/keirf/greaseweazle@flippy-reverse
```

Here is a command line I use to convert a flippy side 1:
```
gw convert a.scp --tracks=hswap --reverse a.d64
```

I suppose the question then is whether this suffices?
1. Do you want to be able to reverse during `gw read` of side 1?
2. Do you want to be able to `gw read` both sides at once, reversing only side 1, and converting/analysing the two sides as two separate D64s?
3. Do you want to be able to *write* flippy disks?

And so on. The basic reverse mechanism was trivial to implement, so it's all about the workflow integration.

klaussoft, 2024-08-21:
Yes, it works on an image I only read side 1.
I tested it with two images, where I read both sides, one with 80 tracks, one with 40 (step=2), but I'm not able to extract side 1.
These are the both files I attached above.

Your questions:

1. It is not necessary, but would be grea, because other tools wourld be able to read the SCP, too
2. It's the same. It's not necessary, but great
3. yes

keirf, 2024-08-21:
> Yes, it works on an image I only read side 1. I tested it with two images, where I read both sides, one with 80 tracks, one with 40 (step=2), but I'm not able to extract side 1. These are the both files I attached above.

I can definitely extract side 1 from your double-sided images. The trick is in the `--tracks=` option:
```
gw convert test_ab.scp test_ab_side1.d64 --tracks=hswap --reverse
gw convert test_ab_full.scp test_ab_full_side1.d64 --tracks=hswap:step=2 --reverse
```

> Your questions:
> 
> 1. It is not necessary, but would be grea, because other tools wourld be able to read the SCP, too
> 2. It's the same. It's not necessary, but great
> 3. yes

Then I think I should add the reverse option to the `--tracks` option. So you could do a raw dump like:
```
gw read --tracks=c=0-39:h1.off=-8:h1.reverse a.scp
```

What do you think?

> Edit: OK, now I was successful with both images. But why I have not to set the head to h1? In the image are datas for h0 and h1.

Because `hswap` swaps the heads, so h1 becomes h0.

klaussoft, 2024-08-21:
> Because hswap swaps the heads, so h1 becomes h0.
But why it doesn't work with --tracks=c=0-39:h=1? That sould be the same result or what's wrong in this thinking?

keirf, 2024-08-21:
> Then I think I should add the reverse option to the --tracks option.

Actually you can do a raw plain old `gw read --tracks=c=0-39:h1.off=-8 a.scp` and then split out and analyse later. For example:
```
gw convert --tracks=h=0 a.scp a_side0.scp
gw convert --tracks=h=0:hswap --reverse a.scp a_side1.scp
```

Now you have split the two sides into separate flux files, with flux in the correct track order for both. You can now use third-party tools, or go through `gw convert` again, as you prefer:
```
gw convert a_side0.scp a_side0.d64
gw convert a_side1.scp a_side1.d64
```

This doesn't give a story on writing, but the sides could be written separately with side1 something like:
```
gw write --tracks=hswap:h0.off=-8 --reverse a_side1.d64
```

This does avoid adding even more complexity to `--tracks=` syntax. I don't know. I suspect most people will pull outless hair having it as a top-level option (`--reverse`) like this.

keirf, 2024-08-21:
> > Because hswap swaps the heads, so h1 becomes h0.
> But why it doesn't work with --tracks=c=0-39:h=1? That should be the same result or what's wrong in this thinking?

Because the format analyser is set up on tracks 0-39 head 0. So data presented as head 1 does not ge passed to the analyser. Hence the need to rename the tracks onto head 0, which is what `hswap` does. So there's an internal logic here, even if it seems arcane and stupid in this case.

klaussoft, 2024-08-21:
> Because the format analyser is set up on tracks 0-39 head 0. So data presented as head 1 does not get passed to the analyser. Hence the need to rename the tracks onto head 0, which is what hswap does. So there's an internal logic here, even if it seems arcane and stupid in this case.

OK, now I understood.

> Actually you can do a raw plain old `gw read --tracks=c=0-39:h1.off=-8 a.scp` and then split out and analyse later. For example:
>
> ```
> gw convert --tracks=h=0 a.scp a_side0.scp
> gw convert [...]

That sounds good.

keirf, 2024-08-22:
I will look into getting this all done and rolled into the next release, 1.19.

closed 2024-08-22 by keirf