keirf/greaseweazle #463: Bad sectors

midi-creator, 2024-07-30:
Greetings, how does greaseweazle work with bad sectors?

Does greaseweazle know how to mark them and how does it write the image if it finds a bad sector marked?

I would like to learn how to write files to bad floppy disks past bad sectors on greaseweazle.

midi-creator, 2024-07-30:
between the bad sectors, like DOS does

keirf, 2024-07-30:
Do you mean how does it write IMG files. for example?

midi-creator, 2024-07-30:
How do I write to a bad floppy disk with bad sectors?

keirf, 2024-07-30:
Why would you want to write to a bad floppy disk?

Greaseweazle completely overwrites all data on existing tracks. So, as long as any sector errors on a disk are *soft* errors, they will get completely overwritten. If your disk has *hard* errors (eg. there is a physical problem with the magnetic media, or the disk is dirty) then that's a separate issue and the disk is best discarded.

midi-creator, 2024-07-30:
Well, because a lot of these floppy disks are in stock. Dos has no problem with this: it writes files between bad sectors

midi-creator, 2024-07-30:
I guess greaseweazle  is not for me ((( I'll have to keep on keeping the old DOS(((( computer.

keirf, 2024-07-30:
Yeah, Greaseweazle has no filesystem awareness, so mapping files around bad sectors is out of scope.

closed 2024-07-30 by keirf