keirf/greaseweazle #464: Guy Spy (Readysoft) (Atari St) IPF fails to write correctly

DaveHarrison1968, 2024-07-30:
Hi Keir, I have an issue when writing IPF images back to floppy disk of "Guy Spy-Atari ST" it's a ReadySoft game. Dflrsilver has made a set of temp IPF's from my original Raw files, for me to test, but they just will not write correctly. He recommended that I send them to you for further investigation.  
I am thinking that it could be related to issue #358 as the whole disk format seems to have sectors crossing the index mark. Here is a link to the IPF's, my original RAW files and a RAW file from IPF of disk 1 that I tried to write to disk.
Many Thanks Dave

https://we.tl/t-OPx5X4OleT

keirf, 2024-07-30:
Yes it's the same issue as #358 and #407 -- probably the IPF is made by SugarConvDsk.

keirf, 2024-07-30:
I have a work-in-progress branch to support these non-standard IPF images. You can try the latest automated build here:
https://github.com/keirf/greaseweazle/actions/runs/10163151191

If you are logged in to GitHub you will see win32 and win64 build artifacts at the bottom of the page linked above. Alternatively if using Linux or Mac you can try the branch by doing something like
```
pipx install git+https://github.com/keirf/greasewazle@sugarconv
```

DaveHarrison1968, 2024-07-30:
Thank you for that build, it worked perfectly for those Guy Spy IPF's. Just to add those IPF images were provided by a SPS representative using the official tools. I did try it on the Kick Off II IPF's, but they do not work yet, but brilliant progress being made.
Thank You again and if you ever need any testing on builds, please keep me in mind, always happy to help.

closed 2024-08-11 by keirf