keirf/greaseweazle #465: The program does not record .ipf and .raw images - how to fix it?

mvbhz, 2024-07-30:
I started testing the Greaseweazle F1 Plus Flux. When I try to burn a .raw file to a floppy disk, I get the message:

** FATAL ERROR:
Bad Kryoflux image name pattern 'C:\games\ps.raw'
Name pattern must be path/to/nameNN.N.raw (N is a digit)

It also won't burn a .ipf floppy image:

IPF Image File:
 SPS ID: None (https://github.com/keirf/disk-utilities)
 Platform: Amiga
 Created: 2023/11/26 19:38:36
 Cyls: 0-83  Heads: 0-1
Writing c=0-79:h=0-1
Format amiga.amigados
T0.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
** FATAL ERROR:
T0.1: 11 missing sectors in input image

How to fix it?

keirf, 2024-07-30:
What sort of a raw file is it? If it's an SPS CTRaw image, try changing the file name to `ps.ctr`.

What is your command line for writing the IPF image file. It should be something like `gw write name.ipf`. Don't specify a `--format`!

mvbhz, 2024-07-31:
I know it's a .raw Amiga disk image file. The page with the file says:

"disks that required Nibble Copy or had read errors. Those have undergone the KryoFlux treatment and are supplied as .raw files. Please consult the [KryoFlux documentation](https://www.kryoflux.com/) how to open them".

When I change the file extension to .ctr it records, but I haven't checked yet if it works on a real Amiga. I'll let you know.

As for the .ipf file, after removing the "--format amiga.amigados" parameter, the file is written to a floppy disk. I'll let you know if it works. Thanks for your help so far.

mvbhz, 2024-08-02:
.ipf and .raw (renamed to .ctr) don't work when I run disks burned from these files on a real Amiga. Regular .adfs work, but e.g. extended .adf doesn't work either.

keirf, 2024-08-03:
CTR is not a good format to write back. The vast majority (nearly all) official IPF images write back fine and with verification. Problems here can be due to needing to boot from internal drive df0, and game incompatibility (chipset, ram, Kickstart, etc) rather than a problem with the disk.

mvbhz, 2024-08-03:
But it wasn't originally a .ctr image, but a .raw, in KryoFlux. Maybe Greaseweazle can't record from these images. I don't blame him.

The game in .ipf works fine in the WinUAE emulator, so it's impossible that this image is corrupted. This is a game for Amiga 500 and unfortunately it doesn't work on a real Amiga. Maybe I need to set some additional parameters in the program?

keirf, 2024-08-03:
Kryoflux raw image files are usually raw dumps and they get written back best effort with no verification. 

They are quite different than IPF image files. What's the command line for your IPF? What IPF is it? What Amiga are you testing it on?

mvbhz, 2024-08-03:
The command line (in GreaseweazleGUI) is default:

gw.exe write "C:\games\Ghosts'n'Goblins (UK) (1991-01) (v3) (Full Release + 512k - 1Mb versions) (Elite Systems - Capcom).ipf"

Here is the .ipf:

https://ftp2.grandis.nu/turran/FTP/TheZone/files/_2023/Ghosts n Goblins (UK) (1991-01) (v3) (Full Release + 512k - 1Mb versions) (Elite Systems - Capcom) [Fixed ipf].zip

I'm testing on Amiga 500, 1MB, Kick 1.3. This is the version of the game exactly for this Amiga.

keirf, 2024-08-03:
I will have a look but GnG can be fussy even about which A500 it will work with. No super Agnus for example. Perhaps try some other IPFs from the official SPS archive and see how you get on?

mvbhz, 2024-08-03:
My Amiga doesn't have Super Agnus. It's a regular version of ECS, with old Agnus and old DENISE. I'd have to look for other .ipf files, but that's when I have some free time.

mvbhz, 2024-08-03:
I recorded it again using GW GUI and the game works! I don't know what caused it. In X-COPY in checkdisk it only shows the numbers 2. So it's correct.

keirf, 2024-08-03:
Problem solved haha

mvbhz, 2024-08-03:
True! I've checked some other .ipf images and they work too. .raw still displays the message "Name pattern must be path/to/nameNN.N.raw (N is a digit)". But .ipf is already a lot!

keirf, 2024-08-03:
The raws may work when renamed to ctr

mvbhz, 2024-08-03:
I've already tried that - it records but the game doesn't start, the Amiga floppy drive "clicks" twice and that's it.

I tried changing the file name to ps00.0.raw but then I get the message:

Writing c=0-81:h=0-1
Traceback (most recent call last):
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python311\site-packages\cx_Freeze\initscripts\__startup__.py", line 141, in run
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python311\site-packages\cx_Freeze\initscripts\console.py", line 25, in run
  File "gw.py", line 3, in <module>
  File "C:\hostedtoolcache\windows\Python\3.11.9\x64\Lib\site-packages\greaseweazle\cli.py", line 69, in main
  File "C:\hostedtoolcache\windows\Python\3.11.9\x64\Lib\site-packages\greaseweazle\tools\write.py", line 253, in main
  File "C:\hostedtoolcache\windows\Python\3.11.9\x64\Lib\site-packages\greaseweazle\tools\util.py", line 329, in with_drive_selected
  File "C:\hostedtoolcache\windows\Python\3.11.9\x64\Lib\site-packages\greaseweazle\tools\write.py", line 254, in <lambda>
  File "C:\hostedtoolcache\windows\Python\3.11.9\x64\Lib\site-packages\greaseweazle\tools\write.py", line 74, in write_from_image
  File "C:\hostedtoolcache\windows\Python\3.11.9\x64\Lib\site-packages\greaseweazle\flux.py", line 179, in flux_for_writeout
IndexError: list index out of range

keirf, 2024-08-03:
It's definitely CTRaw. Feel free to share a non-working one and I can maybe see why it's not working.

mvbhz, 2024-08-03:
https://gofile.io/d/xy8bDa

keirf, 2024-08-03:
The problem is that the tracks aren't index aligned, and sector data crosses the index mark. Since Greaseweazle writes index-to-index (placing the write splice at the index), any sector overlapping the index will be corrupted.

You can see this for yourself using HxC Floppy Software https://hxc2001.com/download/floppy_drive_emulator/HxCFloppyEmulator_soft_beta.zip

Load the raw file (you will need to rename to ctr!) and then use the track analyzer to visualise the track data.

*If* the data is AmigaDOS (no copy protection) you may find that `--format=amiga.amigados` helps for this particular CTRaw image file. But it isn't a general solution.

mvbhz, 2024-08-03:
OK, so what parameters do I need to select in Greaseweazle GUI to make it record?

keirf, 2024-08-03:
I don't use or develop the GUIs, but there's probably a "Format" tick box and then you can probably select a format from a drop-down list.

mvbhz, 2024-08-03:
I started it with the parameter but it doesn't record:

CTRaw Image File:
Cyls: 0-82 Heads: 0-1
Writing c=0-79:h=0-1
Format amiga.amigados
** FATAL ERROR:
T0.0: 1 missing sectors in input image

keirf, 2024-08-03:
Then I think this CTRaw cannot be written; quite possibly it is intriniscally corrupted.

closed 2024-08-04 by keirf