keirf/greaseweazle #467: Driver for windows 7 needed

crestr1, 2024-08-06:
is this available anyplace

keirf, 2024-08-06:
Use zadig to set up usbser driver for Greaseweazle device. And use the win32 build of the tools.

crestr1, 2024-08-07:
Tks K, Any bad experiences of this cohabiting with the driver for kryoflux under w7 ?
Been having lots of unrelated issues with Kyroflux it seems to have internal clock speed preservation problems (pic attached). This is why I'm trying greaseweazle
[image]

keirf, 2024-08-07:
Should be fine. Not sure what those KF errors mean, could be bad disk or drive?

crestr1, 2024-08-08:
Both the disk and drive are good, this happens a lot when recovering from 8" drives
I put the missing data error streams in KF down to meta-stability someplace in its logic when a data edge corresponds with a clocking edge.
eg: https://nandland.com/lesson-13-metastability/

closed 2024-08-11 by keirf