keirf/greaseweazle #471: Feature Request: Add Support for CMD FD 2000 (and optionally CMD FD 4000) images

markusC64, 2024-08-14:
Can you please add support for CMD FD 2000 (and optionally CMD FD 4000) images?

CMD FD 2000 images have the extension .d2m for hd and d1m for dd.
CMD FD 4000 images have the extension .d4m for ed. The drive itself also supports the disks from the FD 2000.

The following config enables flashfloppy to handle HD images:
[cmdfd2000::1658880]
cyls=81
heads=2
file-layout = sides-swapped,interleaved
   secs = 10
   bps = 1024
   id = 1
   mode = mfm
   interleave = 1
   gap2 = 22
   gap3 = 100
   iam = no
   rate = 500
tracks = 0-80.0
   h = 1
tracks = 0-80.1
   h = 0

This should fully specify the d2m format.

d1m has half the rate, half the secs and gap sizes to be specified later.
d4m has double the rate  double the secs and gap sizes to be specified later.

In general it is similar to 1581, but more tracks and larger sectors together with dd, hd and ed support.

markusC64, 2024-08-15:
Here is a sample image:

[cmd-fd2000.7z.zip](https://github.com/user-attachments/files/16624567/cmd-fd2000.7z.zip)

markusC64, 2024-08-15:
[cmd-fd4000.7z.zip](https://github.com/user-attachments/files/16624611/cmd-fd4000.7z.zip)

markusC64, 2024-08-15:
Finally I managed to get the final missing sample image:

[cmd-fd2000-dd-empty.7z.zip](https://github.com/user-attachments/files/16629485/cmd-fd2000-dd-empty.7z.zip)

keirf, 2024-08-20:
Now done. The patch is in master for download. Or Windows prebuilts are here:
https://github.com/keirf/greaseweazle/actions/runs/10475628257

keirf, 2024-08-20:
Please reopen if there are any issues.

closed 2024-08-20 by keirf

markusC64, 2024-08-20:
I am nor sure if we should treat that as a bug.

I observed a IAM inside the output og gw convert some.d2m some.scp

But the format description says: "iam = no".

Hobestly, a real CMD FD should have no problem with a IAM since they are quite often fed with a disk written by a PC controller with "samdisk", and they should have a IAM and work.

But strictly the presence of a IAM is wrong for CMD FD formats.

keirf, 2024-08-20:
I doubt it matters but I can turn that off.

keirf, 2024-08-20:
I wonder whether the 1581 drive writes an IAM?

markusC64, 2024-08-20:
Not sure. We might leave it as is for now.

I am getting unsure about the IAM. If it helps for some controllers reading that disk a IAM might be good.

keirf, 2024-08-20:
Take your pick really. Pretty much nothing cares about *reading* it, or not. Most IBM-compat FDCs will include one in a track format, but not all. I've been looking around for raw 1581 disk dumps and couldn't find any, sadly.

The reason it is included in your conversion here is that I simply didn't include `iam=no` in the format descriptions. But we can fix that. And FlashFloppy too (but there D1M and D81 share the same format descriptor, hence it would be nice to know what is right for 1581).

markusC64, 2024-08-20:
If you really want to know whether the 1581 writes a IAM while formating or not, there is one possibility left: I can use the 1541 Ultimate II+ to find out. It emulates a 1581, and it's controller forwards all commands to a process on the ultimate. There I could log the arguments of the format command and we would see.
So doubt possible, executing the original ROM would tell us every detail.

keirf, 2024-08-20:
That would be interesting. Ultimately what we really want is someone with a 1581 to format a disk with it and then read it with Greaseweazle to raw image file for inspection. But 1581 is a bit rare I think? Is UII+ really an alternative here? I got the impression it really deals with 1581 only at a sector level (D81 <-> serial to C64).

markusC64, 2024-08-20:
Yes it is. With a firmware modification I can log the raw command the floppy sends to the mfm controller.
And a WD 1772 compatible controller wants the whole track as parameter with some reserved values for A4  with missing clock and C2 with missing clock. So you would get the format send to the mfm controller, which must be correct.

keirf, 2024-08-20:
Well that is interesting then! So it would be a case of dumping that buffer and inspecting it. The UII+ runs the 1581 original ROM?

markusC64, 2024-08-20:
The ultimate runs the ROM I want it to run. The original ROM, the jiffydos rom (with speeder) or any compatible ROM I/you want.

keirf, 2024-08-20:
I found a 1581 ROM disassembly and it does not write an IAM.

keirf, 2024-08-20:
There you go, that should get rid of the IAMs :) I will do similar for FlashFloppy.