keirf/greaseweazle #472: Host Tools 1.18 no longer detect the greaseweazle using win7 pro

BeastOldone, 2024-08-17:
Host Tools 1.18 no longer detect the greaseweazle using win7 pro. Detection works using Host Tools 1.17.1 just fine. Also I use the 32 bit versions, since I git a missing api-ms-win-core-path-l1-1-0.dll when using the 64 bit version.

A gw info comand in 1.18 ends with device not found.

While 1.17.1 spits out all the info from com port to firmware version.

keirf, 2024-08-18:
It is correct to use the 32-bit build for Win7. The 64-bit build uses a newer version of Python which is incompatible with Win7.  I do not officially support Win7 now, but keep it limping along for now.

Device discovery method changed in 1.18. Perhaps it is incompatible with Win7. I will ask another user who I know has used Win7. Also, are you happy to try test builds of the tools so that this can be fixed on your system?

BeastOldone, 2024-08-18:
Thanks, for the quick response.
Just to let you know I found a modified version of the missing core_path.dll made for blender which appears to allow the 64bit version to run on win7. My main use of the greaseweazle is for making boot disks for old retro computers I keep around.  And help repair/test old floppy drives. Its just a hobby of mine.

I won't mind trying out test builds on my win7 PC. I do have an old laptop running win10 on which 1.18 works just fine, plugged into a usb2.0 port. Have not tried it on a usb3.0 port.

Walter

keirf, 2024-08-20:
Can you try the automated build (under "Artifacts" at bottom of page) here:
https://github.com/keirf/greaseweazle/actions/runs/10467404069

You will need to be logged in to GitHub to see the build artifacts.

BeastOldone, 2024-08-20:
USB detection is working again. Here is the output from the "info" command. Thanks for the quick fix. I look forward to the official relase.

C:\Program Files\Greaseweazle v4>"C:\Program Files\Greaseweazle v4\greaseweazle\gw.exe" info
*** TEST/PRE-RELEASE: 1.19.dev4+gafaa085
*** Use these tools ONLY for test and development!!
Host Tools: 1.19.dev4+gafaa085
Device:
  Port:     COM2
  Model:    Greaseweazle V4
  MCU:      AT32F403, 144MHz, 224kB SRAM
  Firmware: 1.5
  Serial:   GW135194F24205400002181705
  USB:      Full Speed (12 Mbit/s), 128kB Buffer

keirf, 2024-08-20:
I am hoping to get a second opinion on v1.18 on Win7 tomorrow. If I get a second repro of the original issue I will apply the fix to master and roll out v1.18.1.

BeastOldone, 2024-08-20:
Just to verify, I d/l a fresh copy of the 32 bit version and this is the output from the "info" command. And you can see the exact version number of my win7. Thanks again for you time and effort to fix this.

Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Program Files\Greaseweazle v4>"C:\Program Files\Greaseweazle v4\greaseweazle\gw.exe" info
Host Tools: 1.18
Device:
  Not found

keirf, 2024-08-20:
Thanks for the extra test!

keirf, 2024-08-22:
Hi, I received a proper fix from the author of the new Windows device discovery implementation. It tests okay for me in a Windown 7 VM. Could you give it a whirl? The artifacts will be linked from here:
https://github.com/keirf/greaseweazle/actions/runs/10503565856

closed 2024-08-22 by keirf

sandritus, 2024-08-22:
[image]
Tested on Win7 Pro 32 bit [Version 6.1.7601] with Zadig 2.9 using a usb 2.0 port.

BeastOldone, 2024-08-22:
Installed the new 1.19 32 bit and all is working fine.

Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Program Files\Greaseweazle v4>"C:\Program Files\Greaseweazle v4\greaseweazle\gw.exe" info
Host Tools: 1.19
Device:
  Port:     COM2
  Model:    Greaseweazle V4
  MCU:      AT32F403, 144MHz, 224kB SRAM
  Firmware: 1.5
  Serial:   GW135194F242
  USB:      Full Speed (12 Mbit/s), 128kB Buffer