keirf/greaseweazle #473: Problems reading the B side in Apple II flippy 5.25" disks

xjols, 2024-08-20:
I'm tried to read Apple IIe 140Kb 5,25" 2 sides flippy disks
The main side I can make a directly working directly image with:

`gw read --format apple2.appledos.140 --raw flux\disk1\track00.0.raw --drive=A --dd L --tracks="step=2" `
and:
`gw convert --format apple2.appledos.140 flux\disk1\track00.0.raw disk1.do`

This makes a .do 140KB correct working image of the A side. The problem is with the botom side.
I tried read both sides in raw with:
`gw read Flippy_v1.scp --drive=A --tracks="c=0-35:step=2" --dd L --adjust-speed=300rpm`

and then open the .scp raw data with HxC and with "edit tools" tried to **"swap sides"** and **"Reverse Tracks"**. Exporting now the data as ".do" makes a 136KB image thats not works in emulator.

It's possible to **add the opion of reverse flux data in convert options?**

Thanks very much for you hard and soft. It's amaizing!

keirf, 2024-08-20:
This is the Apple2 version of issue #460 

Can you supply a flux image file or two?

xjols, 2024-08-20:
Yes I see that is the same problem with the Commodore flippy disks in Apple II.

The original top side ripped with `gw read --format apple2.appledos.140 --raw flux\disk1\track00.0.raw --drive=A --dd L --tracks="step=2" `
[disk1.zip](https://github.com/user-attachments/files/16681453/disk1.zip)

The two sides raw ripped with `gw read Flippy_v1.scp --drive=A --tracks="c=0-35:step=2" --dd L --adjust-speed=300rpm`
[Apple_04-A-SIMULADOR_MILLON-B-DEFENDER_35tr_h0_h1_300rpm_ddL.zip](https://github.com/user-attachments/files/16681483/Apple_04-A-SIMULADOR_MILLON-B-DEFENDER_35tr_h0_h1_300rpm_ddL.zip)
Another one:
[Apple_03-A-COMPILADOR-B-COPION_36tr_h0_h1_300rpm_ddL.zip](https://github.com/user-attachments/files/16681503/Apple_03-A-COMPILADOR-B-COPION_36tr_h0_h1_300rpm_ddL.zip)

It seems in HxC that the bottom side have 5 tracks offset. But unfurtainly my drive don't accept negative values for offset in h1:
```
gw read Apple_08-A-PASCAL-B-FISICA_35tr_h0_h1.off-8_300rpm_ddL.scp --drive=A --tracks="c=0-34:step=2:h1.off=-8" --dd L --adjust-speed=300rpm
Reading c=0-34:h=0-1:h1.off=-8:step=2 revs=3
Command Failed: Seek: Invalid cylinder -8
```
 I suspect that the only solution will be open the jacket and swap the floppy ?

Thank you very much

keirf, 2024-08-20:
Do you know if your drive is "flippy modded" with a track0 sensor override on pin 33? That is the original mod by the Kryoflux authors and Greaseweazle won't allow stepping to negative cylinders except via this sensor override method (which it autodetects).

xjols, 2024-08-21:
My drive is an  Chynon FZ-506 not modded. Don't spin without index pulses.
Requires hole pulses to work.

keirf, 2024-08-21:
Sorry, now I understand. Indeed you have a few options for flippy reads:
1. Modded drive that can step to negative tracks
2. Modded drive with an extra index sensor (https://learn.adafruit.com/flippy-floppy-drive-modification?view=all)
3. Flip in jacket, or jacket with two index holes

keirf, 2024-08-21:
I have a new `gw convert` option for you to try. This is on branch `flippy-reverse`. You can find Windows build here (you need to be logged in to GitHub to see the Artifact links):
https://github.com/keirf/greaseweazle/actions/runs/10487456321

Or if on Mac or Linux you can try installing with pipx:
```
pipx install git+https://github.com/keirf/greaseweazle@flippy-reverse
```

Here is the command line I use to convert your flippy side 1 (as best as possible since you are missing 4 tracks that your drive cannot access!):
```
gw convert Apple_04-A-SIMULADOR_MILLON-B-DEFENDER_35tr_h0_h1_300rpm_ddL.scp --tracks=hswap:h0.off=-4 --reverse a.do
```

xjols, 2024-08-21:
Hello! Tested your improvment in test relese. Worked as expected. Makes a correct Apple II ".do" image with the data of the bottom side with flux reversed. As you said, the raw data loses the first 4 tracks in my drive.
It would be very usefull for people that have a mooded head Drive that can read tracts 0-4 of the bottom side on flippy disks.
Thank you very much for you work and speed to create this release