keirf/greaseweazle #475: clarify "a standard 34-pin ribbon cable"

fommil, 2024-08-22:
I ordered what I thought would be "a standard 34 pin ribbon cable" (I'm planning to connect to a [Teac FD-505](https://gekk.info/misc/Teac%20FD505%20Dual%20Function%20Combo%20Diskette.pdf) of dubious quality that I found on ebay)

https://www.amazon.co.uk/dp/B0CQHJPHNL

but the cable I bought is far too miniature for both the greaseweazle 4.1 (that just arrived today from your official store, thank you), and the Teac. What kind of cable should I be buying? Everything I see looks like the same size as the one I bought, not the size that fits the greaseweazle.

In addition, if you have any idea what kind of power supply I'm supposed to use for the Teac, then I'd be all ears because the manual just says "the power connector consists of the 4 separate wires attached to a single connector" but it's much smaller than the 5.5mm x 2.1mm Molex connector that I bought on ebay.

keirf, 2024-08-22:
Post a picture of the rear of your drive, as I think some have 3.5-inch style connector and some have 5.25-inch. These are different sorts of connectors, both data and power.

Standard 34-pin ribbon is 2.54mm pitch. You can literally just buy a "floppy cable" on eBay. You can get ones with both 3.5-inch and 5.25-inch connections on them. Like https://www.ebay.co.uk/itm/226286144745

For power if it's a 5.25-inch connector you need a Molex connector, else you additionally need a Molex-to-floppy converter. Things like https://www.ebay.co.uk/itm/335446676852 and https://www.ebay.co.uk/itm/126026461719 (just examples, not recommending these listings over others).

keirf, 2024-08-22:
Power your drive directly from an external PSU such as above. Do not use the power connector on the Greaseweazle. Do not change jumper settings on the Greaseweazle.

fommil, 2024-08-22:
Yeah I completely wasted my money on the Molex cable too. I bought this at a ridiculous markup to make my life easier, blindly following the greaseweazle docs (before I received the Teag) https://www.ebay.co.uk/itm/374339467971 so that I could just plug it in to a DC power supply but it's completely the wrong shape, as you can see here... the power supply on the Teag looks exactly the same shape as the cable on the greaseweazle. So I guess I actually need a "4 Pin Floppy Drive Power Connector" on one side and a female DC plug on the other. But why 4 pins? Is it supplying 5V and 12V with independent grounds or something? I don't want to have to build something to bridge it. The Teag docs say it needs 5V AND 12V which I find very confusing.

Here's a pic of the rear end of the device, it matches the docs I linked in the original issue description

[image]

fommil, 2024-08-22:
I note that the "missing pin" is in the center middle on the Teag. But the missing pin is 3 in from the outside on the greaseweazle. Does that mean I should be getting a cable with a twist?

fommil, 2024-08-22:
oh right, I see about the power supply now. I mistakenly thought the "molex" was the name of the floppy end. But it's the name of what would have been the PC. The specialist PSU is supplying the 5V / 12V mixture... so my little cable thing is probably only supplying the 5V side and was a complete waste of money. Thank you @keirf for the links, I should have gotten that in the first place.

I'm starting to regret getting the Teag. It seems everything is much simpler with a single 5V 3.5 floppy that can be powered from the greaseweasle (although I would still need to buy the floppy-floppy power cable).

It's a real shame that fully constructed devices aren't available on the store, I would have bought one at a premium without thinking twice, and probably saved myself a fortune once wasted cash on useless cables was taken into account.

keirf, 2024-08-22:
Unusual missing pin on the TEAC. But don't worry, you can basically get any floppy ribbon cable straight or twisted. That will only switch which drive (3.5 vs 5.25) is A vs B. You don't need a ribbon with 5.25-style edge connectors. Something like this would do: https://www.ebay.co.uk/itm/155503438788

Actually 5.25-inch drives usually do have a "Molex connector". But this drive has a 3.5-inch style connector. All you need is a simple adapter such as I linked in an earlier post.

The sad thing is that the 3.5-inch slim drive on these TEACs is usually broken anyway. It's super common. So you may well end up getting another 3.5-inch drive anyway. And that will probably be 5v only, and could probably be powered direct from the Greaseweazle using the two smaller connectors on an adapter such as https://www.ebay.co.uk/itm/190602613674

Getting a good working drive is half the battle. I know I have better things to do than source and sell them!

keirf, 2024-08-22:
PS. If the 5.25-inch part of your TEAC works then you can consider that a win as working 5.25 inch drives are far harder to find than working 3.5 inch drives, these days. And they're way more expensive.

fommil, 2024-08-22:
@keirf thank you! I don't know what possessed me to buy the Teag. I don't even own any 5.25 disks, I just thought "well if I'm investing time into finally getting this to work, I may as well get them both working so I can be the local retro gaming recovery guy". I might just get a 3.5 as you suggest and be done with it. All I really wanted to do was recover my old SWOS save games from 30 years ago 😆

keirf, 2024-08-22:
I can probably spare you a working drive and some cables for a reasonable cost. Email me if interested. Maybe you want to test the Teac first.

closed 2024-08-23 by keirf

fommil, 2024-09-03:
Thanks all, all the new parts arrived and I was able to scan and old Amiga disk to ADF and now have my full catalog to go 😄 this is awesome, I've wanted to do this for 30 years.

The teag is only able to recover 99% of the disk so I think it's junk. The 5.25 might still work but I have no need for it. If any of you wants it for parts or as a repair job, I'm happy to ship it to you in the UK for free... otherwise it's headed for landfill.

keirf, 2024-09-03:
I'll certainly take a look, thanks! I sent you an email with my details.