keirf/greaseweazle #476: Reading Thomson TO-DO disks

fplanque, 2024-08-23:
I am new to greaseweazle. 
I was able to make images of Atari ST disks to confirm it works.

But my main goal is to read disks generated by the Thomson MO5, MO6, TO7, TO7/70, TO8, TO8D, TO9, TO9+ line of computers in the 1980ies.

More specifically, I have first generation 5.25" disks, I'd like to read.

According to an old manual, these have the following specs:
- Single sided
- 40 tracks
- 16 sectors per track
- 128 bytes per sector
- (disk capacity 80 kB)

They also called it "format IBM 16 sectors/track" in the manual. 

I don't think there is any such format supported by `gw`. The smallest I see is `ibm.160`. Am I mistaken?

Are the formats hardcoded in `gw` or is there a configuration file to add more?

keirf, 2024-08-23:
There is a configuration file to add more. See the default file here: https://github.com/keirf/greaseweazle/blob/master/src/greaseweazle/data/diskdefs.cfg

You can define a new diskdefs config file and pass it via the `--diskdefs=` option.

You can also try dumping these disks using the `ibm.scan` format, which will scrape FM and MFM sectors at the usual bit rates. 

All the above makes sense for dumping to IMG format. You can also dump raw flux images without any format conversion at all.

fplanque, 2024-08-23:
Thank you @keirf . 

I looked at this example in the file:
```
disk ibm.160
    cyls = 40
    heads = 1
    tracks * ibm.mfm
        secs = 8
        bps = 512
        gap3 = 84
        rate = 250
    end
end
```

I understand how to tweak it a little:
```
disk thomson.80
    cyls = 40
    heads = 1
    tracks * ibm.mfm
        secs = 16
        bps = 128
        gap3 = 84
        rate = 250
    end
end
```

But I really don't understand what `gap3` and `rate` mean.

Is there some documentation I could read to understand this?

Thank you.

fplanque, 2024-08-23:
oh, I see some comments at the top of the file ;)

keirf, 2024-08-23:
You don't usually need to specify any gap values. Rate can often be automatically reduced by gw too.

fplanque, 2024-08-23:
Ok great.

For anyone who comes here, I was able to read a disk with this cfg:
```
disk thomson.scan
    cyls = 40
    heads = 1
    tracks * ibm.scan
        # Following options restrict the scanner's search
        # rate = 250
        rpm = 300
    end
end
```

Probably a good idea to set cyls to 40 to avoid horrible noises that probably don't do the drive any good ;)

I am not yet sure what to do with the .img and if it is the same as the thomson emulators .fd, but it's a start:
```
T38.0: IBM FM (16/16 sectors) from Raw Flux (75571 flux in 400.36ms)
T39.0: IBM FM (16/16 sectors) from Raw Flux (76959 flux in 400.36ms)
Cyl-> 0         1         2         3         
H. S: 0123456789012345678901234567890123456789
0. 0: ........................................
0. 1: ........................................
0. 2: ........................................
0. 3: ...................X....................
0. 4: ...................X....................
0. 5: ........................................
0. 6: ........................................
0. 7: ...X....................................
0. 8: ........................................
0. 9: ........................................
0.10: ........................................
0.11: ........................................
0.12: ...X....................................
0.13: ...X....................................
0.14: ........................................
0.15: ........................................
Found 635 sectors of 640 (99%)
```

closed 2024-08-26 by keirf