keirf/greaseweazle #477: Missing sectors and IMG format

fplanque, 2024-08-23:
I realize have 2 reads of the same floppy that yielded different missing sectors:

```
Cyl-> 0         1         2         3         
H. S: 0123456789012345678901234567890123456789
0. 0: ........................................
0. 1: ........................................
0. 2: ........................................
0. 3: ...................X....................
0. 4: ...................X....................
0. 5: ........................................
0. 6: ........................................
0. 7: ...X....................................
0. 8: ........................................
0. 9: ........................................
0.10: ........................................
0.11: ........................................
0.12: ...X....................................
0.13: ...X....................................
0.14: ........................................
0.15: ........................................
Found 635 sectors of 640 (99%)
```

vs

```
Cyl-> 0         1         2         3         
H. S: 0123456789012345678901234567890123456789
0. 0: ........................................
0. 1: ...................X....................
0. 2: ........................................
0. 3: ...X....................................
0. 4: ........................................
0. 5: ........................................
0. 6: ........................................
0. 7: ........................................
0. 8: ........................................
0. 9: ........................................
0.10: ........................................
0.11: ........................................
0.12: ...................X....................
0.13: ...X....................................
0.14: ........................................
0.15: ........................................
Found 636 sectors of 640 (99%)
```

On track 19, one image has sector 4 and the other has sector 12.

So, my question is: is there a tool to combine the 2 images to get a more complete one?

Also, it seems to me that the .img file size varies depending on missing sectors. But the sectors are not numbered inside the image file, are they?

So dropping a sector screws up anything that comes after it, doesn't it?

Is there a way for `gw` to write missing sectors as 0x00 instead of ignoring them?

keirf, 2024-08-24:
Interesting that you find different missing sectors on each read. Also I would expect `gw` to be including bad sectors in the output IMG file.

Anyhow, you really want a proper diskdef for this format, rather than relying on `ibm.scan`. If you attach a raw dump (`gw read --tracks=c=0-39 name.scp`) we can have a look and make a precise diskdef. But something like:
```
disk thomson
    cyls = 40
    heads = 1
    tracks * ibm.fm
        secs = 16
        bps = 128
        rate = 125
    end
end
```

This will guarantee you a correctly laid out IMG file, and also will let you write an IMG file back to disk (because theis diskdef describes the track layout, while `ibm.scan` does not). It will also fill bad sectors with a fixed pattern which later scripts could use to merge two IMG files containing different bad sectors.

fplanque, 2024-08-24:
Indeed, this is what I have been working with in between:

```
disk thomson.80
    cyls = 40
    heads = 1
    tracks * ibm.fm
        secs = 16
        bps = 128
        #gap3 = ??
        #rate = ???
	rpm = 300
    end
end
```

(rpm I found in the tech specs)

And it does produce all .img files with the same size. 

Actually after a closer look, the only size difference is for a "system disk" where the entire track 0 is missing (maybe on purpose) and in that case ibm.scan did not write it to the .img).

> `rate = 125`

Is this also necessary for writing? Did you guesstimate it or did you calculate it?

I will also test with Thomson 3.5" disks (but for now I have the big bulky 5.25" drive with "open heart surgery" connected)

Some of my disks have private data (mini database files) so I need to triple check before attaching an image here.

>  It will also fill bad sectors with a fixed pattern which later scripts could use to merge two IMG files containing different bad sectors.

Do these scripts exist yet?

If not, what is the pattern? (I might give it a try)

Would you recommend a different format (.imd ?) for the case

When renaming the .img files to .fd, I can use them directly in an emulator such as DCMOTO. Is there a way to write the file extension directly as .fd (but same content as .img) to avoid tedious file renames?

Thank you.

fplanque, 2024-08-24:
PS:

> Interesting that you find different missing sectors on each read. 

In the overwhelming majority of tries the bad sectors are the same. But on rare occasions, I get a variation. Maybe when I reinserted the disk it was ever so slightly centered differently that it made a difference on the weak tracks.

Now I am also going to try in hotter and colder rooms (betting on different dilatation of disk vs drive), as every recovered sector matters to me ;)

keirf, 2024-08-24:
The `rate = 125` is standard data rate for FM (single density) on a micro-diskette (as opposed to the original 8-inch floppy disks). It's rarely deviated from.

The script doesn't exist. The pattern is `-=[BAD SECTOR]=-` repeated to fill the bad sector. So pretty easy to detect!

Yes I can support FD format. It would be interesting what the default format should be, and also behaviour for double-sided formats (should the file order be interleaved cyl0h0, cyl0h1, cyl1h0, cyl1h1, ... or should it be all of head 0 and then all of head 1). I think Thomson supports a range of disk formats across the series?

keirf, 2024-08-24:
Oh, and IMD is a nice format, if your other tools/emulators support it! I suppose you can always transcode to IMG/FD later.

keirf, 2024-08-24:
> Now I am also going to try in hotter and colder rooms (betting on different dilatation of disk vs drive), as every recovered sector matters to me ;)

Have you inspected the disk media? Or try visualising it in HxC software (https://hxc2001.com/download/floppy_drive_emulator/HxCFloppyEmulator_soft_beta.zip):
1. Take a raw dump, eg. to SCP image file
2. Load it into HxC software
3. Click on "track analyzer"

Now you can look per-track at "stream view" which shows you raw flux timings, and you can see how noisy your data looks.

You can also look at "disk view" and look for patterns of bad sectors (orange) or bad flux timings (red).

fplanque, 2024-08-24:
> Oh, and IMD is a nice format, if your other tools/emulators support it! I suppose you can always transcode to IMG/FD later.

No, the Thomson emulators don't support it, that's why I started with the simple .img.

> Yes I can support FD format. It would be interesting what the default format should be, and also behaviour for double-sided formats (should the file order be interleaved cyl0h0, cyl0h1, cyl1h0, cyl1h1, ... or should it be all of head 0 and then all of head 1). 

For a DCMOTO .fd file:
- The order of the sectors in the .fd file should be the logical order (no intervleaving)
- The file should contain the entire side 0 followed by the entire side 1 (when double side disk) (in my case I only used single sided)
   - So it should be cyl0h0, ..., cyl79h0, cyl0h1, ... (I think double sided exists only on 3.5" 80 track disks)
   - Fun trivia: DCMOTO can support additional floppy sides concatenated after 0 and 1. In this case they show up as additional disks on the same controller.

What about the .img format?  

- is it cyl0h0, cyl0h1, cyl1h0, cyl1h1? 
- does it have the sectors in physical (interleaved) or logical order?

**^ I think I really need to know that before trying out the 3.5" disks.**

> I think Thomson supports a range of disk formats across the series?

My understanding is that there are 3 possible formats:
- 5.25" SS 80 kB
- 3.5" SS 320 kB (not tested yet)
- 3.5" DS 640 kB (not tested yet)

keirf, 2024-08-24:
IMG format is interleaved. I can add FD as a sequential format with default format SS 80kB (for example).

fplanque, 2024-08-24:
> IMG format is interleaved. I can add FD as a sequential format with default format SS 80kB (for example).

By all means, please do! 👍 
 
> Obviously it's not critical until you test DS disks.

I will do very soon because my end goal is to transfer data from 5.25" disks to 3.5" disks.

fplanque, 2024-08-24:
@keirf In order to save my sanity in additional tests, can you please confirm the following:

1. The `interleave` param is ignored in reading and writing with an `.img` file.
2. The `interleave` param will be crucial in reordering the sectors for an `.fd` file.

An also a question:

- are the sectors numbered on the disk (some magic of "soft sectoring" maybe?) that would allow `gw` to automatically collect all sectors in order, no matter the interleave that was used for a particular disk?

keirf, 2024-08-24:
That's correct, IMG files are always in logical sector order, which is ascending order of sector id (every sector has a header incorporating an id). Interleave (and cskew, and hskew) dictate physical arrangement of sectors on disk, but that's not represented in an IMG file. It is in other formats such as IMD though.

keirf, 2024-08-24:
You can work out interleave by looking at tracks in HxC track analyzer. Simply count the number of sectors between any pair of sectors with consecutive sector IDs. Then add one. Hence interleave of 1 mean sectors are arranged in consecutive ascending order.

fplanque, 2024-08-25:
> That's correct, IMG files are always in logical sector order, which is ascending order of sector id (every sector has a header incorporating an id). Interleave (and cskew, and hskew) dictate [...]

I am confused because you say "that's correct" and then I believe you explain the opposite of what I had in mind ;)

So let me ask this to clarify: if a track on disk has the sectors in this order: `1-3-5-7-2-4-6-8` , will the the **.img** file also have the sectors in the order `1-3-5-7-2-4-6-8` or will it have the sectors ordered as `1-2-3-4-5-6-7-8` ?

Thank you.

fplanque, 2024-08-25:
Following up on:

> You can work out interleave by looking at tracks in HxC track analyzer. 

And also:

> Have you inspected the disk media? Or try visualising it in HxC software (https://hxc2001.com/download/floppy_drive_emulator/HxCFloppyEmulator_soft_beta.zip):
>
> 1. Take a raw dump, eg. to SCP image [...]

I got this:
[image]

This seems to confirm that sectors have an interleave of 1 on this disk but I don't think it's standard on Thomson disks. (So I guess I got lucky for the first tests as there is no way to have them in a wrong order but the question in the previous message above is still paramount to my further testing.)

I also see the [BAD SECTOR] pattern ;)

However, I don't understand where/how I can see noisy data (?)

> You can also look at "disk view" and look for patterns of bad sectors (orange) or bad flux timings (red).

I get this:

[image]

This disk has a least one bad sector (show above, track 27, sector 16, head 0) but it shows up green on this view....

Finally, I am wondering: 
- you recommended .def (which pads bad sectors with ==[BAD SECTOR]==, but is there another format that will have the original corrupted data from the bad sectors (even if the CRC is invalid)? 
- also for .img (and .fd): wouldn't it make sense to have an option to get as much corrupted data as possible into the file? For example, if the sector is from a text based format, even if we have 2 or 3 bytes corrupted but the rest is good, it's valuable.

keirf, 2024-08-25:
> I am confused because you say "that's correct" and then I believe you explain the opposite of what I had in mind ;)
>
> So let me ask this to clarify: if a track on disk has the sectors in this order: [...]

 `1-2-3-4-5-6-7-8`

keirf, 2024-08-25:
> This disk has a least one bad sector (show above, track 27, sector 16, head 0) but it shows up green on this view....

You haven ot done a raw SCP dump because you specified a `--format` to `gw read`. In this case by default the output is always re-encoded from the GW analyser. This means all perfect sectors have been emitted, with "bad sector" data filler for bad sectors. You need to additionally specify `--raw` to `gw read` to get the raw flux emitted. Do that and then see what the HxC visualisation looks like!

fplanque, 2024-08-25:
Thank you. I am learning so much here. 

I feel this is all not documented yet. Can I contribute to the Wiki?

keirf, 2024-08-25:
The wiki isn't open for public edits but I'm happy to take suggestions and contributions and curate them into the wiki.

fplanque, 2024-08-25:
Here's an example with --raw SCP:

[image]

[image]

How bad is the "noisiness" of this? (I don't have any "clean" floppies to compare with)

I guess the black pixels in stream view should be as clustured as possible on only 2 levels representing 0 and 1 ?

Sidenote: now I can observe an interleave factor of 4. Thank you :) Also I did some additional research and it seems Thomson used 4 by default in their first "OS", then 5, and then 7 in the most recent versions. I guess they has their reasons to optimize this. So i'll use 7 by default for writing back to disk with gw.

I am also still researching all possible geometries 5"vs3", SS vs DS, SD vs DD.

keirf, 2024-08-26:
You want the bit bands as tight and close to horizontal as possible. The noise at three locations in the above track view is pretty bad and probably unrecoverable except by inspecting media and perhaps you will find dirt/mould which can be cleaned, if you are lucky.

closed 2024-08-26 by keirf