keirf/greaseweazle #484: Linux cmd interface needed

crestr1, 2024-09-08:
Have just set up the Linux version from the download and all the diagnostic stuff seems to work but i cannot find a command utility that lets me CMD  talk to the greaseweazle board directly from a terminal.   nothing in my system will make "sudo picocom /dev/ttyACM0" work or "sudo stty -F /dev/ttyACM0 9600" get to the board even after "sudo chmod -R a+rwx /dev/ttyACM0" should clear the field. Maybe there is a trick I need to know for this approach.

I have a windows 10 setup with the GUI and the GUI offers a direct CMD interface that allows one to directly talk to the board thru command lines. 
The windows GUI system works brilliantly recovering good disks (IBM FM 128 byte sectors, 77 tracks, SS and DS disks just using its IBM scan mode alone but i need to make complete intact .img files that just have a dummy value placed in bad sectors so the images are full sized and OS directory entries point to the right offset in the .img file when bad sectors are encountered and replaced rather the have their space omitted from the .img

crestr1, 2024-09-09:
I'm recovering 8" disks, these are a bit tricky and the disks do not center perfectly (its 40 year old hardware) and just reloading a disk can make different sectors bad and good, I figure if a get good reads of every sector of a disk in a series of .img files i can grab all the good stuff to make a perfect image of a disk using original os utilities.

crestr1, 2024-09-09:
what happens, terminal seems attached bit is inoperative:
"ubman@ubman-Venus-series:~/Greaseweazel/greaseweazle-1.19/src/greaseweazle/tools$ sudo picocom /dev/greaseweazle
picocom v3.1

port is        : /dev/greaseweazle
flowcontrol    : none
baudrate is    : 9600
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready
"
terminal is inoperative but showing ready

crestr1, 2024-09-09:
installation port scan:
Port 33:
    device: '/dev/ttyACM0'
    name: 'ttyACM0'
    hwid: 'USB VID:PID=1209:4D69 SER=GW00156854541300000541A71B LOCATION=1-1.2:1.0'
    manufacturer: 'Keir Fraser'
    product: 'Greaseweazle'
    vid: 1209
    pid: 4d69
    location: '1-1.2:1.0'
    serial_number: 'GW00156854541300000541A71B'

crestr1, 2024-09-09:
As an aside --- can i recover a disk by have its image created by overwriting an .img of a freshly formatted disk with the good sectors it finds on the disk it is reading that should finally present an identical image size to what it is overwriting.  this can help automate replacement of bad sectors with good freshly recovered ones

crestr1, 2024-09-09:
Some more from my Linux (Mint) system:
ubman@ubman-Venus-series:~/Greaseweazel/greaseweazle-1.19/src/greaseweazle/tools$ gw info
Host Tools: 1.19
Device:
  Port:     /dev/ttyACM0
  Model:    Greaseweazle V4
  MCU:      AT32F415, 144MHz, 32kB SRAM
  Firmware: 1.4
  Serial:   GW00156854541300000541A71B
  USB:      Full Speed (12 Mbit/s)

*** New firmware version 1.5 is available
To perform an Update:
 - Run "gw update" to download and install latest firmware
ubman@ubman-Venus-series:~/Greaseweazel/greaseweazle-1.19/src/greaseweazle/tools$ gw update
Downloading latest firmware: greaseweazle-firmware-1.5.upd
Updating Main Firmware to version 1.5...
** FATAL ERROR:
Could not reopen port after mode switch
ubman@ubman-Venus-series:~/Greaseweazel/greaseweazle-1.19/src/greaseweazle/tools$

crestr1, 2024-09-09:
**The update finally worked after a few re-try's  and the gw utility is available now so I'm OK and underway.**
would appreciate a quick look at the stuff I've marked as_an_aside this feature would be great for all the 8" stuff I've be rescuing from family archives of 60 years. Got over 1000 8" disks and some almost unused drives.
the important thing is I need to make complete intact .img files that just have a dummy value placed in bad sectors so the images are full sized and OS directory entries point to the right offset in the .img file when bad sectors are encountered and replaced rather than have their space omitted from the .img   - some pointers on how to do this would be a big help

keirf, 2024-09-09:
Hi there,

Your idea of inserting/injecting good sectors is pretty neat. But anyway, you can get correct-sized IMG files every time already, with explicit `-=[BAD SECTOR]=-` filler in bad sectors, by explicitly specifying the disk layout. This may need you to write yuour own diskdef. That would be a local file containing a stanza like in https://github.com/keirf/greaseweazle/blob/master/src/greaseweazle/data/diskdefs.cfg

You can then specify your own diskdefs file to gw using the `--diskdefs=` command-line option. This option may not be available from GUI. But then, you can probably handle a command line :)

You can't directly talk to Greaseweazle via a terminal program. It's not *really* a serial interface. It's a lie, to make it easier to interface to Python via pyserial. Properly I should make Greaseweazle a winusb-style device with its own protocol driveable via a library such as libusb. But that's more hassle.

crestr1, 2024-09-12:
Thanks Kier, I'm underway but I'm having trouble understanding how to configure double sided disks. I had a perfect reading using the ibm scan mode. but when i try to alter dec.rx01 to two sided 77 track 128 byte sectors i get troubles:
what do i need to change here to make it double sided to read the same disk as in scan mode does if the disk is error free

disk dec.rx01
    cyls = 77
    heads = 1
    tracks * ibm.fm
        secs = 26
        bps = 128
        gap3 = 26
        rate = 250
        rpm = 360
    end
end

If i make the 1 into 2 for double sided 
it gives sector errors on side one, it is expecting something that is not reading
I need to tell it how side 1 sectors are marked i think

is there an output format i can use in IBM.scan mode that helps me figure this out
Ive attached a file that shows scan mode reads both sides and dec.rx01 reads side 0 perfectly
[Ibm scan mode works.txt](https://github.com/user-attachments/files/16979897/Ibm.scan.mode.works.txt)

keirf, 2024-09-12:
What error do you get for side 1? Or can you share a raw dump?

keirf, 2024-09-12:
Are they really DEC RX01 as I thought they were strictly single sided?

Anyway, you can try the attached diskdef.

[diskdefs.cfg.zip](https://github.com/user-attachments/files/16981737/diskdefs.cfg.zip)

crestr1, 2024-09-12:
Thanks Kier, they are not DEC RX01 i was using RX01 as it was something i thought would work with the disks I've got if I made it two sided.   Is the DEC part of the name attached to some background settings or can i just rename this Moto MDOS3 without consequences. They are 8" disks.  From the first IBM format soft sector systems ever. Have been using kryoflux and HXC formats (.hfe) the attached zip contains an image file and its hfe
[Images_and_hfe.zip](https://github.com/user-attachments/files/16986042/Images_and_hfe.zip)

crestr1, 2024-09-13:
Thanks Kier, sorry to be a pain
This new one works only on some  disks, some have a different format on side 1 apparently.
need to wrap my head around greaseweasel's trickey possibly undocumented bits.
some questions on the magic numbers in the new diskdefs you made for me:
tracks 0-76.1 ibm.fm
    secs = 26
    bps = 128
    gap3 = 26
    rate = 250
    rpm = 360
    id = 27        where in the py code ?: is what this does tabulated anyplace else ?
    h = 0           what does this do.   Loop back to side zero for next pass ? where in *.py
  end
end
point me to py bits if you think this can provide better explanations.

keirf, 2024-09-13:
The ibm diskdef rules are explained here https://github.com/keirf/greaseweazle/blob/master/src/greaseweazle/data/diskdefs.cfg#L10

They are parsed by this class (and this is the whole file dealing with IBM format tracks): https://github.com/keirf/greaseweazle/blob/master/src/greaseweazle/codec/ibm/ibm.py#L802

closed 2024-09-18 by keirf