keirf/greaseweazle #485: Hard-sectored Altair/IMSAI 8-inch floppies

davepl, 2024-09-09:
I've been unable to find a format option that will successfully read and write Altair 8-inch floppies.   They're hard-sectored, 32 sectors, 76 tracks, one side, 128 byte sectors I believe.

Is this something that the greaseweazle can currently do?  If so, would love to know the right arguments, etc, as I've been unable to to make much progress.

Ideally I'd like to be able to write the Altair DSK images that are on the web, but for now I'm content to try reading and writing discs I've previously formatted in the Altair... but can't get that far!

Thanks!
Dave

davepl, 2024-09-09:
I can read an image with:
gw read --raw --tracks=c=0-76:h=0 --revs=5 rx50_blank00.0.raw 

But writing it back always fails with flux underflow.  Direct USB connection on MacOS, no hub.
dave@Daves-Mac-Pro ~ % gw write --tracks=c=0-76:h=0 rx50_blank00.0.raw 
Writing c=0-76:h=0
T0.0: Writing Track (Flux: 166.3ms period, 183.0 ms total, Terminate at index)
Command Failed: GetFluxStatus: Flux Underflow

keirf, 2024-09-09:
I'm intrigued how you are getting 166ms-period raw dumps with the specified command line. Or is this one of the original hard-sector-aware drives (SA801 style) which separate the index and sector pulses onto different output pins?

Please share a flux file. Also please share output from `gw bandwidth`.

keirf, 2024-09-16:
Any further updates for this one?

keirf, 2024-09-23:
Please re-open to continue the discussion.

closed 2024-09-23 by keirf