keirf/greaseweazle #488: non-sequent sector numbers - how-to?

hanz45ru, 2024-09-16:
As I see in diskdefs.cfg, there's possibility to define non-sequent sector numbers. But, when I try to put something like it, into custom format description, i get "invalid literal for int() with base 10". Is t possible only for selected disk encodings, not for any?? Or i do something wrong? Below is my try:
disk zx.isdos.800
    cyls = 80
    heads = 2
    tracks * ibm.mfm
        secs = 1,2,3,4,9
        bps = 1024
	h = 0
    end
end

keirf, 2024-09-16:
Currently it's not possible, except by using the generic format `ibm.scan`.

hanz45ru, 2024-09-16:
this perfectly works for read, but if I want to write IMG, sector nums probably will be in sequence 1-5. So, if not possible, then only solution is HFE format.

keirf, 2024-09-16:
You could use a verifiable format like ImageDisk (IMD)?

I could probably also allow `id` to be specified as a list in the diskdef config.

hanz45ru, 2024-09-17:
let me explain.

keirf, 2024-09-17:
IMD doesn't store gap lengths. You should try it. For example:
```
gw read --format=ibm.scan name.imd
gw convert --format=ibm.scan name.imd name.img
gw convert name.imd name.hfe
gw write name.imd
```
So you can read to IMD, then create IMG and "perfect" HFE. And use the IMD to write back too. The need for `--format=ibm.scan` on the second command line above is a bit of a shame. I'll see about fixing that to make it automatic.