keirf/greaseweazle #489: add real analog capture.

cm68, 2024-09-16:
every floppy disk has a test pin used to perform drive alignment.   this pin is connected to an oscilloscope to see the raw, analog output of the head amplifiers.   this analog signal is processed by the drive's filters and edge detection circuits to generate the digital flux pulses seen at the read data lines captured by the greaseweazel.

the cpus on recent GW all have decent analog to digital converters that are unused, and these could capture much of the raw signal at a high enough rate to allow data recovery of even seriously degraded disks using fairly simple signal processing code.

work required would be to wire an ADC enabled pin to a header, to define an output format, and capture the data correlatable with the flux data already captured.

USB bandwidth will be an issue, as the 12Mbit speed won't cut it.

keirf, 2024-09-16:
Probably would need an MCU with High Speed USB. AT32F405 is interesting (cheap) but may be insufficient grunt and ram for serving a High Speed interface. Probably back to STM32F730Z8 as used on Greaseweazle F7 Lightning Plus.

cm68, 2024-09-17:
the STM32F730Z8 has a pretty potent ADC subsystem, which can be driven by DMA.  if I read the spec right, it can get 6 bit samples at 12.5 mhz, which should be plenty to get a decent waveform, but this will take some very gnarly chip setup; triple interleave reduced resolution with DMA is probably a not very well tested case.

philpem, 2024-09-20:
A better option might be to pair GW off with an oscilloscope. That'd "just" be a software change to the GW software to control the motor, set up the acquisition on the scope, trigger the capture, and move on.

I've done this (manually) with an Agilent MSO-X3000A and it works quite well. The only "tricky" part is the signal processing afterwards.

If you're interested in this, you might find the collaboration I did with Chris "scarybeasts" Evans interesting: https://www.youtube.com/watch?v=QAEWmQ1REeE
We recovered the source code to Repton 4 from a damaged disc, using a combination of analog recovery and speeding up the spindle motor.

keirf, 2024-09-20:
Yes it did strike me that otherwise what we are essentially building is a crappy DSO.

cm68, 2024-09-20:
True.

Building a crappy DSO is "out of scope", pun intended.

A decent amount of writeup on the process, along with software support for
this, would go a long way to making this feature moot.

My specific problem is that I have a lot of damaged disks, and a lot of
good drives.

 I have built a set of drives that are intentionally radially misaligned
and capture my disks 3 times, each on track, and 2 a quarter track off emin
each direction.  I am able to read disks written with misaligned originals.

What I can't recover is weak magnetic domains, or contamination with high
frequency noise that the bandpass filter doesn't catch.  That's where
signal processing might help.

Clearly, this road has been mapped by people before me.  I need to do some
reading.