keirf/greaseweazle #491: Erase only erases side 0

philpem, 2024-09-20:
**When I erase a double sided disc, only side 0 is erased.**

To erase the disc I used the command: ` gw erase --drive B --tracks c=0-79:h=0,1`
Then I read the disc back as an SCP with: `gw read --drive B temp_erase_check.scp`.

I found that the tracks on side 1 had not been erased:

```
gw read --drive B x__ibmpc-525hd.scp
Reading c=0-81:h=0-1 revs=3
T0.0: Raw Flux (60761 flux in 630.76ms)
T0.1: Raw Flux (162332 flux in 778.60ms)
T1.0: Raw Flux (51280 flux in 711.66ms)
T1.1: Raw Flux (145024 flux in 779.97ms)
T2.0: Raw Flux (72760 flux in 726.32ms)
T2.1: Raw Flux (153208 flux in 776.96ms)
T3.0: Raw Flux (51723 flux in 738.53ms)
T3.1: Raw Flux (138950 flux in 782.30ms)
T4.0: Raw Flux (67191 flux in 728.29ms)
T4.1: Raw Flux (145146 flux in 766.91ms)
T5.0: Raw Flux (50355 flux in 718.46ms)
T5.1: Raw Flux (141780 flux in 782.99ms)
```

I can work around this by running the erase command twice: once for side 0, and once for side 1.

**Hardware set up**

Drive: Mitsubishi model M4853-362M - 5.25in 80-track double-sided, NAMFG PCB

Greaseweazle info:
```
$ gw info
Host Tools: 1.19
Device:
  Port:     /dev/ttyACM0
  Model:    Greaseweazle F7 Lightning Plus
  MCU:      216MHz, 256kB SRAM
  Firmware: 1.5
  Serial:   GW004D003D3338510631323733
  USB:      High Speed (480 Mbit/s), 128kB Buffer

$ gw delays
Select Delay: 10us
Step Delay: 10000us
Settle Time: 15ms
Motor Delay: 750ms
Watchdog: 10000ms
```

keirf, 2024-09-20:
I just tested this on an AmigaDOS disk. Seems to work! Maybe this is an artifact of your particular drive somehow? Also, it may be that although side 1 has "more flux" in your read -- is it actually anything other than a different density of noise?

```
viper:greaseweazle$ gw read a.adf
*** TEST/PRE-RELEASE: 1.20.dev3+g21d0f5a
*** Use these tools ONLY for test and development!!
Reading c=0-79:h=0-1 revs=1.1
Format amiga.amigados
T0.0: AmigaDOS (11/11 sectors) from Raw Flux (43800 flux in 219.57ms)
T0.1: AmigaDOS (11/11 sectors) from Raw Flux (42860 flux in 219.57ms)
T1.0: AmigaDOS (11/11 sectors) from Raw Flux (42844 flux in 219.57ms)
T1.1: AmigaDOS (11/11 sectors) from Raw Flux (43127 flux in 219.57ms)
T2.0: AmigaDOS (11/11 sectors) from Raw Flux (43063 flux in 219.57ms)
T2.1: AmigaDOS (11/11 sectors) from Raw Flux (43232 flux in 219.57ms)
T3.0: AmigaDOS (11/11 sectors) from Raw Flux (43452 flux in 219.57ms)
T3.1: AmigaDOS (11/11 sectors) from Raw Flux (43092 flux in 219.57ms)
T4.0: AmigaDOS (11/11 sectors) from Raw Flux (55406 flux in 219.57ms)
T4.1: AmigaDOS (11/11 sectors) from Raw Flux (55559 flux in 219.57ms)
T5.0: AmigaDOS (11/11 sectors) from Raw Flux (55573 flux in 219.57ms)
^C
viper:greaseweazle$ gw erase --tracks=c=0-3:h=0,1
*** TEST/PRE-RELEASE: 1.20.dev3+g21d0f5a
*** Use these tools ONLY for test and development!!
Erasing c=0-3:h=0-1, revs=1
T0.0: Erasing Track
T0.1: Erasing Track
T1.0: Erasing Track
T1.1: Erasing Track
T2.0: Erasing Track
T2.1: Erasing Track
T3.0: Erasing Track
T3.1: Erasing Track
viper:greaseweazle$ gw read a.adf
*** TEST/PRE-RELEASE: 1.20.dev3+g21d0f5a
*** Use these tools ONLY for test and development!!

Reading c=0-79:h=0-1 revs=1.1
Format amiga.amigados
T0.0: AmigaDOS (0/11 sectors) from Raw Flux (26718 flux in 219.57ms)
T0.0: AmigaDOS (0/11 sectors) from Raw Flux (110210 flux in 900.11ms) (Retry #1.1)
T0.0: AmigaDOS (0/11 sectors) from Raw Flux (193611 flux in 1579.56ms) (Retry #1.2)
T0.0: AmigaDOS (0/11 sectors) from Raw Flux (276806 flux in 2261.97ms) (Retry #1.3)
T0.0: Giving up: 11 sectors missing
T0.1: AmigaDOS (0/11 sectors) from Raw Flux (25155 flux in 219.56ms)
T0.1: AmigaDOS (0/11 sectors) from Raw Flux (95990 flux in 838.91ms) (Retry #1.1)
T0.1: AmigaDOS (0/11 sectors) from Raw Flux (174973 flux in 1530.79ms) (Retry #1.2)
T0.1: AmigaDOS (0/11 sectors) from Raw Flux (253247 flux in 2214.68ms) (Retry #1.3)
T0.1: Giving up: 11 sectors missing
T1.0: AmigaDOS (0/11 sectors) from Raw Flux (26721 flux in 219.57ms)
T1.0: AmigaDOS (0/11 sectors) from Raw Flux (123710 flux in 1012.92ms) (Retry #1.1)
T1.0: AmigaDOS (0/11 sectors) from Raw Flux (205487 flux in 1686.17ms) (Retry #1.2)
T1.0: AmigaDOS (0/11 sectors) from Raw Flux (288915 flux in 2372.73ms) (Retry #1.3)
T1.0: Giving up: 11 sectors missing
T1.1: AmigaDOS (0/11 sectors) from Raw Flux (24658 flux in 219.57ms)
T1.1: AmigaDOS (0/11 sectors) from Raw Flux (93552 flux in 841.55ms) (Retry #1.1)
T1.1: AmigaDOS (0/11 sectors) from Raw Flux (170642 flux in 1539.39ms) (Retry #1.2)
T1.1: AmigaDOS (0/11 sectors) from Raw Flux (246947 flux in 2228.67ms) (Retry #1.3)
T1.1: Giving up: 11 sectors missing
T2.0: AmigaDOS (0/11 sectors) from Raw Flux (27073 flux in 219.57ms)
^C
```

philpem, 2024-09-21:
It could be an artefact of the drive - without looking at the waveforms the GW is putting out it's hard to tell.

I do have the datasheet for the drive - http://www.bitsavers.org/pdf/mitsubishi/floppy/M4852_M4853/M4852_M4853_Specifications.pdf

This quotes a track-to-track time of 3ms, a settling time of 15ms, a head loading time of 50ms, and a motor start time of 250ms.

There is a warning about changing the side select:

> Side one select.
> This interface line is used to select which of the two sides of the diskette should be read or written. When this line is
at logical "1," the Side 0 head is selected; or when it is at logical "0," the Side 1 head is selected. If the polarity of the side one select signal is reversed, delay read/write operation by more than 100 us before execution

But this is probably more relevant:

> Upon completion of a write operation, reverse the polarity of the side one select signal after a delay of 1000 us. The heads are tunnel erase type, with a physical core gap deviation between the read/write head and the erase heads so with no delay, non-erased areas would be generated on the diskette due to a timing difference between the write data
area and the erase area during write operation. This is prevented by delaying the erase current cutoff time of a few hundred microseconds within the M4852/M4853. Therefore, the head select must not be reversed during this delay time. Also, the track access action must not be permitted for 1000 us.

How would I configure this constraint on the Greaseweazle?

keirf, 2024-09-21:
That's a good point, there probably should be seek-to-write and write-to-seek delays.

However, `gw erase` is actually overrunning each erased track by 10% (approx 20ms) so the problem raised in the datasheet shouldn't affect you (https://github.com/keirf/greaseweazle/blob/master/src/greaseweazle/tools/erase.py#L36).

If you wanted to linger after write or erase, the simplest hack to try would be a `time.sleep(0.01)` in `usb.write_track()` and `usb.erase_track()` (for example here: https://github.com/keirf/greaseweazle/blob/master/src/greaseweazle/tools/erase.py#L36)

philpem, 2024-09-22:
I added a delay after the write command (line 36, indented outside the "if" block) as you suggested, which fixed the problem.

I also tried adding a delay after "usb.seek()" (line 29, erase.py) and sure enough, it erased both sides. This probably isn't as correct a fix if the tunnel-head shutoff delay is the issue.

You also asked about the difference in flux --

> Also, it may be that although side 1 has "more flux" in your read -- is it actually anything other than a different density of noise?

I see this on a lot of drives, discs, and reading tools (the DiscFerret sees it too). The noise density is usually different - and side 1 usually has more fluxes recorded for erased tracks. I've never been able to find a good explanation, but it's definitely just a difference in noise.

Here's side 0, where the noise seems to sort itself into bands:
[image]

And side 1, where it seems to coalesce into the bottom 4us:
[image]

keirf, 2024-09-22:
So what problem is solved by adding the post-write delay? Since the original problem appeared to be more fluxes reported on side 1 after erase. But the remainder of your comment gives an explanation of sorts for this, and kind of classifies it as a non-gw issue. So what change is effected by the (admittedly correct) post-write delay?

philpem, 2024-09-22:
> Since the original problem appeared to be more fluxes reported on side 1 after erase

The problem was side 1 wasn't being erased at all. It had about 140,000 fluxes -- which is what a normal data stream looks like.

I've just written a SCP file back to the disc (`gw write --drive B image.scp`) which gave a flux plot like this in HXC when I read it back:
[image]

After an erase (`gw erase --drive B`) without the patch I expected both sides to be blank, but only Side 0 is (aside from a few stray tracks):
[image]

After adding the delay (in either location) and erasing again, both sides are erased:
[image]

The problem is exactly what I put in the issue summary - "Erase only erases side 0".

philpem, 2024-09-22:
I've just read back what I wrote - the drive does have a difference in noise density on erased tracks which i think is normal. But the noise tracks are still only around 60,000 fluxes at most.

A track with actual, valid data is around 120,000 fluxes or more - sorry I wasn't clear on that.

keirf, 2024-09-22:
Ah ok. You could have proved it with a decoding read command to show that side 1 does still have valid data. But anyway I believe you. Most drives seem to have minimum delays before/after writes but this drive has a slightly longer post-write delay specified and seems to somehow enforce it. Like, the second write-gate assertion gets ignored completely. I wonder if back-to-back writes on the same track are allowed? I can't think of a great reason for such a thing though.

philpem, 2024-09-22:
I think the PDF is pretty clear about the reasoning - the head is a tunnel erase type which places the erase gaps behind the write head (rotationally speaking), so it needs to stay on for a little longer after the end of a sector to trim the guard bands.

The lockout circuit seems to be the unusual part - both the amount of time and that it locks out a succeeding write operation.

I expect the sector gaps would mean you wouldn't need to do back-to-back writes - by my reckoning one revolution at 300rpm is 200 milliseconds. The timeout is 1000us = 1ms, which should be absorbed by the intersector gap time.

keirf, 2024-09-22:
Okay well I will fix this in firmware and with two extra configurable delays, which I will call pre- and post-delay, defaulting to 100us and 1000us resp. I will have firmware for you to try quite soon.

keirf, 2024-09-22:
I have implemented a fix here: https://github.com/keirf/greaseweazle-firmware/commit/099bf362f05e0b36c5cecf079673b02c796b1ee7

There is an automated build here: https://github.com/keirf/greaseweazle-firmware/actions/runs/10982254136

Please can you download the Artifact at the bottom of the above page and unzip it, then test the new firmware with no hacks to the host tools:
```
gw update --force --file path/to/update_file
```

keirf, 2024-09-23:
Depends what you're decoding, I agree. By the way you can always try different PLL parameters with disk-analyse. `--pll-period-adj` and `--pll-phase-adj` variations will generally get you as good or better than HxC. The defaults are 5 and 60 respectively, which is quite aggressive. Good alternatives for a unform-rate track with high-frequency noise/dirt are phase=1 and period=10.

PS. If you can decode with `gw`, even better! As this will by default use 5/60 falling back to 1/10, so you get the best of aggressive and conservative out of the box. These settings are also overrideable on the command line, of course.

closed 2024-09-23 by keirf