keirf/greaseweazle #495: GW Floppy Disk drive alignment uses

crestr1, 2024-10-02:
Hi Keir,
Floppy Disk drive alignment uses of Greaseweazle.
Do you have any utilities for GW to support the disk head alignment adjusting. I have alignment disks from various makers that have track pairs adjacent to tk40 or 44 these are normally used looking at an oscilloscope but because the L and R patterns are sine-wave based i thought you my have some clever stuff that can work out the incoming wave peaks in a GW and report when the L&R alternate peaks are the same level for alignment purposes.
There are also other tests on alignment disks for track zero etc, If you have code that supports any of this I would greatly appreciate being able to use it..

_Originally posted by @crestr1 in https://github.com/keirf/greaseweazle/issues/487#issuecomment-2378134927_

keirf, 2024-10-02:
These alignment disks require use of analog test points on the drive, to which gw doesn't have access. I do plan to implement a much simpler alignment test which will simply report the current sectors that are readable, in a continual read-ans-report loop. Very useful but in no way as precise as a proper analog test disk.

crestr1, 2024-10-06:
My gut feeling on this is because of the way the alignment track is constructed
perfect alignment (reading the alignment track normally) would be a true flux-null since the alignment off-track sine waves would be out of phase and cancel. misalignment maybe would produce outputs but at what state of misalignment this would occur depends on the head quantization levels for the zero and one discrimination. We need some R&D here.
 I'd suggest all we need is a GW setup routine that parks the read head on the alignment track and looks results of  single  track seeks in and out which generally yield a slight under-seek misalignment on the "TO" track after single track seeks. Long seeks generally have the slight misalignment in the over-seek direction because of greater inertia.
These are effects i came to use writing disk driver error recovery software in the 1970's for WD floppy controller chips when they first arrived on the market.

crestr1, 2024-10-06:
I attach some DYSAN Digital diagnostic disk info:
these were a diagnostic aid for technicians encountering disk reading problems in the 1970's
I have an unused 800-001 that will probably need an oven cycle and 24 Hour Cool and acclimatization before use. It is only single sided.:

[Dysan DDD Brochure.pdf](https://github.com/user-attachments/files/17267918/Dysan.DDD.Brochure.pdf)
[Dysan Digital Diagnostic Disk.pdf](https://github.com/user-attachments/files/17267919/Dysan.Digital.Diagnostic.Disk.pdf)

screwtop, 2024-11-04:
It would be great to have functionality for aligning drives without an alignment disk or oscilloscope too. I recently aligned a couple of 3.5" drives using Greaseweazle hardware and software and a small set of original commercial diskettes: I'd rotate the track stepper during reading to find the limits of readability, and then fix it at the midpoint. I used some shell one-liners to read the same track repeatedly, but it occurred to me that a `gw ... --loop <n>` option would be handy for this (and perhaps more generally). Values of <n> less than 1 could mean continuous.

Adjusting the track 0 sensor was trickier. I used a pad of small post-it-style flags as an adjustable spacer between the sensor PCB and the drive frame - each leaf is about 0.09 mm or about half a track, which seemed to be adequate resolution. A mode for stepping back and forth continuously between two specified tracks (it's not always 0 and 1, apparently) reporting the track 0 sensor state (without re-zeroing?) would help here. Perhaps another use case for a `--loop` option?

crestr1, 2024-11-04:
using the DDD alignment disk (see specs above) I get the following indication of the ability to reliably read +- 10 thou misalignment on my 8" disk recovery system:
[DDD Disk Read.txt](https://github.com/user-attachments/files/17624888/DDD.Disk.Read.txt)
A track zero sensor setup that cycles a restore and a track 1 seek would be useful to use setting/checking the T0 sensor. On some 8" drives there is no adjustment if it is wrong you need to replace the sensor module.

screwtop, 2024-11-05:
With the 3.5" DD drives, I found I could adjust the radial head alignment about +/- 0.03 mm (about 1.2 thou) before read errors occurred. I believe the track width on these is about 0.115 mm (track pitch is 0.1875 mm).

Thanks for posting the Dysan PDFs. When I was getting started I'd found the Dymek guide, which was incredibly helpful for understanding the principles, even if the alignment disks themselves are now virtually unobtainable!

crestr1, 2024-11-05:
what brand and model 3.5 drive are you using ST
my 8" is YE data YD180 - 1601 this is a half height DS 8". (has 50 way connector converter to GW)

screwtop, 2024-11-07:
I've mostly been working on Amiga drives: some Chinon FZ-354 and Panasonic JU-253-043P units, a Mitsumi D357T2, and a Chinon FZ-357 that was easy to modify for Amiga use. Most were working just fine as they were, but it was one of the Panasonics that got me started on the alignment track (pardon the pun). I think there was terrible backlash in the leadscrew due to hardened grease, and it was skipping steps. After cleaning, lubricating and aligning I ended up with the stepper pretty much exactly where it was from the factory!

At work we have a Mitsubishi M2896-63-02U 8" drive which shows signs of life on the bench power supply but we'll need a 50-pin adapter to use it.

crestr1, 2024-11-09:
that Mitsubishi M2896-63-02U 8" is very similar to my YE data
the 50 way converter is available on eBay for about $30, i think it comes from France, I have several of them,

screwtop, 2024-12-20:
@crestr1, you might be able to help. I got a 50-way converter and fixed some problems with this Mitsubishi 8" drive (shorted capacitor and an open inductor on the +24 V line). Drive spins, stepper moves, and I now see index pulses. However, I don't know what format the disks are or how to work that out systematically. With no other information, I guess a reasonable assumption would be 77 tracks, 26 sectors per track, 128 bytes per sector? I don't really want to risk messing with the alignment or TK00 sensor without good reason, but it's possible that will be necessary. I do not have an 8" alignment diskette. Thanks!

crestr1, 2024-12-20:
gw has a mode that detects IBM formats, you just let it loose and it will tell you
But the basic native IBM format is as per your reasonable assumption 77,26,128/
It will of course expect that it can access the drive and head loads happen when requested
it it will need the basics: drive select, tr 0 and all controls for stepping and data lines and no spurious connections for special modes in the drive
So start with IBM mode read, it will tell you if it cannot access the drive from there you will need to suss out the appropriate connections if no drive select

crestr1, 2024-12-20:
The blown cap and inductor would suggest a reversed 24V voltage was applied, I would expect it may have resulted in a failure of a driver device for a phase of the stepper control or head load solenoid if there is no reverse voltage protection diode in place that took the destructive load and fused the inductor

screwtop, 2025-01-15:
Greatly appreciate the advice, thank you, especially the head cleaning method. Good news: the drive is working! I borrowed a commercial diskette with some software for CP/M so I could be confident that it had data in a known format. Decent looking signals from the head test points, and all sectors readable in the expected format. It will be good to try with a double-sided disk to confirm that both heads work correctly, but I'm very happy to have got it to this point!