keirf/greaseweazle #497: Question:Using multiple greaseweazle's at once

4w50m3d3v516n3r, 2024-10-06:
Is it possible to use multiple greaseweazle's at once, to backup multiple disks at once? I have tried using a raspberry pi and the latest greaseweazle firmware and cannot get it to work permanently. I started each "gw" instance as a separate background process, assigning the respective USB device. It works from the start, but then at least one disk-drive will stop working. If I do it in a sequential manner, it works all the time.

Here is my bash-script:

`#! /bin/bash

gw read --format=ibm.1440  test.img --device /dev/ttyACM0  &

gw read --format=ibm.1440  test1.img --device /dev/ttyACM1 &

gw read --format=ibm.1440  test2.img --device /dev/ttyACM2 &

gw read --format=ibm.1440  test3.img --device /dev/ttyACM3 &

wait < <(jobs -p)`

Is something like this possible? If so, I would like know how. 

Thank you.

keirf, 2024-10-06:
How does it stop working? With a flux overflow error?

4w50m3d3v516n3r, 2024-10-10:
I think it is some kind of weird USB Issue on the Raspberry PI 3. Even if the boards and the drives are powered externally I got the following dmesg warning:
`WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 7`
I am not an USB expert. Searching on Google for this message and trying the fixes did not help. Some say it is because of the 64bit Version of the raspi-os, some say, you have to make changes within config.txt.
Anyways , after setting up an old PC with lubuntu and setting up the greaseweazel again, everything works as expected.

4w50m3d3v516n3r, 2024-10-21:
I close that thing now.

closed 2024-10-21 by 4w50m3d3v516n3r