keirf/greaseweazle #498: Feature request: Make it possible to mount filesystems through gw

mattarroz, 2024-10-12:
First of all, thanks for this great project! I'm using the greaseweazle to debug a broken Atari ST floppy drive.

However, I wonder if it is possible to write individual files to a floppy using the greaseweaszle. I know I can mount a filesystem image, copy files to it, unmount it and then rewrite the image to a floppy. But is it really necessary to write an entire image? Can't I just write individual files?

Under linux, this could be realized with a greaseweazle kernel driver, with a combination of mtools+gw or maybe by using pyfuse3+gw, I guess.

keirf, 2024-10-12:
There is a tool called disk flashback for windows, by Rob Smith. Noone has implemented this for Mac or Linux however.

mattarroz, 2024-10-13:
Here is a rough sketch of how this could work:

[fusefatfs](https://github.com/virtualsquare/fusefatfs/blob/master/fatfs/source/00readme.txt) and DiskFlashback are both based on the fatfs library. DiskFlashback uses Dokany as a drive virtualization mechanism, whereas fusefatfs is using FUSE, comptabile with many platforms (Linux, MacOS and even Windows). So by adapting Rob Smith's [GreaseWeazleInterface.cpp](https://github.com/RobSmithDev/FloppyDriveBridge/blob/main/floppybridge/GreaseWeazleInterface.cpp) to interface with [fusefatfs.c](https://github.com/virtualsquare/fusefatfs/blob/master/fusefatfs.c), this could potentially work... at least for FAT based file systems.

I don't have time to try this very soon, unfortunately.

keirf, 2024-10-13:
Definitely it could work, as DiskFlashBack is based on a Windows version of FUSE. We could consider Rob's project a curated set of filesystem backends. However I too am unlikely to take this on any time soon.

mattarroz, 2024-10-13:
Even though I had more than enough TODOs this afternoon, I could not resist hacking something together:
https://github.com/mattarroz/gwmount

I used fusefatfs as a basis, but replaced the `disk_read, disk_write`, etc functions declared in [diskio.h](https://github.com/mattarroz/gwmount/blob/b0b6e55e9553179e9b57ffd108baa8d67cbe81f1/fatfs/source/diskio.h#L30) with the [implementations](https://github.com/mattarroz/gwmount/blob/b0b6e55e9553179e9b57ffd108baa8d67cbe81f1/DiskFlashback/MountedVolumes.cpp#L18) provided by Rob's DiskFlashback. There are some places where he is using windows APIs which needed equivalent POSIX/Linux implementations. In the future, maybe these fixes could be merged into DiskFlashback using `#ifdef`s for linux or so. Then I linked his floppybridge that directly talks to the Greaseweasle. The entry from fusefatfs to DiskFlashback is [here](https://github.com/mattarroz/gwmount/blob/b0b6e55e9553179e9b57ffd108baa8d67cbe81f1/fusefatfs/fusefatfs.c#L418). Not sure about that.

I did not test it yet, however it compiles and links successfully. There may be errors in my quick and dirty approach, anyone should feel free to fix or improve the code.

With pyfuse3 then maybe one could use shared libraries built from Diskflashback to provide a `gw mount` command. Alternatively one could create a python interface for fusefatfs.

GhostDog98, 2025-10-07:
Is there any update on this? Considering making a grease weasel but would probably only seriously consider it if this was a somewhat reliable feature...