keirf/greaseweazle #501: Greaseweazle writes floppies correctly (?) but on real hardware (Amiga) there are errors

mvbhz, 2024-10-21:
Hello

I had a situation several times when Greaseweazle wrote a disk from an .adf file but no error was displayed. After inserting this disk into a real Amiga and checking in X-COPY there are errors on the disk (6 and 8). After changing to a working disk it is OK. It seems that the program and/or the device does not verify the correctness of the write. Is it possible to set Greaseweazle in such a way that when the disk is damaged the program verifies errors on the disk?

My equipment is Greaseweazle F1 Blue Pill. The program I use to write is Greaseweazle 1.20 + GreaseweazleGUI.

keirf, 2024-10-21:
Writes are verified. You should see "all tracks verified" at the end of the write. Possibly one of your drives is slightly out of alignment. You could also try reading one of these disks back to adf on Greaseweazle after observing an issue on Amiga, and see whether Greaseweazle can read it or not.

mvbhz, 2024-10-21:
The message "all tracks verified" appears every time, even with damaged disks.

It is probably not out of adjustment, because I have already checked 3 different disk drives. Besides, if the heads were set incorrectly, there would be no errors on the WinUAE emulator in X-COPY, but there are errors.

The same thing happens when formatting a diskette. It looks as if the Greaseweazle 1.20 program formatted without verification (the formatting speed is high, as if without verification). Additionally, I started the G-COPY program and it already recognizes errors when formatting.

keirf, 2024-10-21:
It does verify. What command do you execute to "format" a disk?

mvbhz, 2024-10-21:
The command that is executed is:

"C:\Amiga\Greaseweazle\greaseweazle-1.20\gw.exe" erase

If there is any other command to format a diskette with verification, let me know.

keirf, 2024-10-21:
In short, if you "gw write" an ADF from the command line it will be verified unless you explicitly specify --no-verify. If you have concerns over GreaseweazleGUI behaviour please raise them with the gui's maintainer.

closed 2024-10-21 by keirf

mvbhz, 2024-10-21:
But this is not a GUI issue. Without the GUI I type "gw.exe erase" in cmd and it is exactly the same. It formats but does not see errors. On the same device on G-COPY or in the WinUAE emulator under X-COPY the errors are visible.